TMRh20/TMRpcm #17: 16-bit Mono on Arduino Nano

mmame, 2015-02-12:
First of all, many thanks for the great TMRpcm library. It works on 8-bit very well on Arduino UNO!

I also tried the 16-bit advanced sample from git to play 16bit Mono (using a 1:256 resistor ladder) but unfortunately, on the MSB output pin the sound is heavily "clipped" - it sound like overamplified (on the LSB pin output I get "noise" as supposed). I tried different sampling rates (12~32kHz), but no success. Any idea?

TMRh20, 2015-02-12:
Hi, thanks!

The only thing that comes to mind immediately is that you may need to lower the volume setting. 

I would suggest also taking a look at the 16-bit circuits [here](http://www.hackerspace-ffm.de/wiki/index.php?title=SimpleSDAudio#Audio_output_connection) for reference

The above link is also to another audio library for Arduino that focuses more on performance and quality over simplicity. If you are looking for higher quality audio, it may be worth taking a look.

mmame, 2015-02-12:
I generated a 32kHz 16-bit Mono file -3db @1kHz from this site http://www.audiocheck.net/audiofrequencysignalgenerator_sinetone.php and looked with an oscilloscope (please ignore the mismatching RC lowpass...). The result looks weird - but wait! The sine is just there, but looks like mirrored on a half of period - imho, the tmrpcm interprets the data as unsigned - but most pcm 16bit mono files are signed, aren't they?
[image]

TMRh20, 2015-02-13:
Haha, yes that would make a lot of sense, now that I see the oscilliscope output. That would also mean the library has been handling 16-bit files incorrectly this whole time. 

Theoretically, I think a fix would be easy enough, by converting the samples to an unsigned format before feeding to the timers. I put updated TMRpcm.cpp code into a github gist [here](https://gist.github.com/TMRh20/bdfa1306bc57c8fc1f48) but haven't been able to test it out yet. 

Feel free to try it out if you have a chance.

mmame, 2015-02-13:
Hi!
Wow that was fast. I tried the gist and would like to say "almost done"... :->

Now, the signal looks a it should [image]
 but every ~504ms, I get some strange popping noise (pic2). Maybe my UNO (or the SD card) is just too slow to manage 16bit and a buffer runs out

[image]

. Anyway, I'll do some further investigation an will come back to you.

TMRh20, 2015-02-19:
In taking a bit of a look at this, I noticed some popping noise as well, but it wasn't quite so frequent, and didn't seem to happen with every track. There may be some issues with the volume control when in 16-bit, but otherwise, if it is a read-speed issue, using the sdfat lib instead of the stock sd library might improve performance.

mmame, 2015-02-21:
Back from holidays - sorry for the delay :-)

It is definitely a performance issue: when running at 22kHz or lower sampling rate, 16 bit performs well on a 16MHz Arduino UNO board. At 32KHz, the described popping noise occurs. I will try to slightly overclock my UNO to 22~24MHz which should be fast enough to run 16bit on 32kHz ;-)

Btw, please check Line #673 -> imho, we have a 16 bit signed number - in other words, adding 127 to the 8 lower bits is not required.

closed 2015-02-21 by mmame

jps2067, 2017-05-23:
Hi mmame, 

Actualy i`m working on audio proyect and arduino and ansking how can use that library as you comment before, just read and send for a external DAC.

In my case i would like to use a SPI DAC 8 bit to reduce RAM and processor consume.

Can you share some data to how achieve that easily?

Thanks in advance!!