TMRh20/TMRpcm #21: demo music

ElectricRCAircraftGuy, 2015-04-22:
Hi, it would be great advertising for the library if you posted a section in your wiki with links to demo music coming from an Arduino running your code. Ex: post a youtube video (better), or even an mp3 download (ok), of various songs (with vocals) playing. I'd like to know how well this library can reproduce regular mp3 tracks with vocals, without having to build the circuit and load the code (though I will *eventually* get around to this.

Thanks!
Looks like a great lib.

TMRh20, 2015-04-23:
Thanks for the input. Theres lots of examples on youtube etc, so this would not be too dificult. For now, here are some potential links to use. Feel free to add any more.

http://www.instructables.com/id/Playing-Wave-file-using-arduino/

http://timewitharduino.blogspot.com/2013_05_01_archive.html

http://www.instructables.com/id/Hockey-Goal-Horn/

https://www.youtube.com/watch?v=zaoVO1optkA

https://www.youtube.com/watch?v=XgKXwTjAgXM

Technically the library will only play .wav files, mp3 is too much to handle.

gamecompiler, 2017-01-29:
I will record some of them next weekend. Shouldnt need that much time.
Perfect chance to use my new audio breadboard adapter 😈

gamecompiler, 2017-01-30:
Coud you please give me push access? Ive got some demo music ready. I also recorded it from the arduino to give people a demo of what Arduino pcm sounds like.

gamecompiler, 2017-03-31:
Music has been added, closing now.

closed 2017-03-31 by gamecompiler