TMRh20/TMRpcm #23: question about recording

michaelynx, 2015-05-28:
hello, i have a little question about the recording mode, I want to record a signal that's little ( 1V peak to peak) and i would prefer not to amplify it ( if possible) before putting it into the arduino. so my signal will theoretically oscillate between 2 and 3 volts before going to the arduino, does the tmrpcm library notice it and uses the 10 bit resolution between 2 and 3 volts are does he uses the 10 bits resolution between 0 and 5 volts ? if so is there a way to make it only take the voltage between 2 and 3 volts ?

i hope my question is clear and thanks to anyone who can help me 

have a nice day

michaelynx, 2015-05-28:
now that i think about it i think it's an 8 bit resolution not 10, but the question remains the same :)

mmame, 2015-05-29:
Well, you could set the ADC Reference Voltage to the maximum Voltage you
want to record (e.g. 3V by using a resistor divider). But as I know, the
minimum Voltage (AGND) is always == GND so you might need to add a "Virtual
GND".

TMRh20, 2015-05-29:
You could also try change the internal reference to 1.1v ( 2.5v on Mega boards) by changing
```
ADMUX |= _BV(REFS0) | _BV(ADLAR);
```
to
```
ADMUX |= _BV(REFS0) | _BV(REFS1) | _BV(ADLAR);// Analog 5v reference, left-shift result so only high byte needs to be read
```

See http://arduino.cc/en/Reference/analogReference and http://www.atmel.com/dyn/resources/prod_documents/doc8161.pdf

closed 2015-10-01 by TMRh20