TMRh20/TMRpcm #24: Support For 1284P

JackTrow, 2015-06-19:
The 1284P is an Arduino supported microcontroller which I've been using for sometime.

It's shown among other supported controllers here, http://playground.arduino.cc/Main/ArduinoOnOtherAtmelChips

I've downloaded your library and got the SD working however nothing seems to happen when I try to play something.

Check out the pin out with pin functions and can you confirm which pin I should be using. I've tried using D15 as its a PWN output. Should I try any other pins? Does it need to be on a timer pin? If yes, which pin would that be? Some explanation would go a huge way.

Pinout : http://blog.protoneer.co.nz/arduino-pinout/ (scroll down)

Thank you, looks like a absolutely fantastic library that I will be using in my product if I can get it working.

JackTrow, 2015-06-19:
I'm using the "basic" sketch, the code runs however I got nothing from any of the PWM pins, have you got any suggestions or ideas?

This library is like gold if it works on the 1284P.

TMRh20, 2015-06-20:
On 1284P the Arduino Pins to use are 13 &12 ( the timer1 pins OCR1A and OCR1B)
Set speakerPin to 13
Set pin 12 to output

Timer3 support can be added for pins 6,7 also, but without modification, it should work as is with timer1.

Please let me know if this works.

JackTrow, 2015-06-21:
Thanks for the response!!

I'm using my scope to measure the output and I connected as you said but no output signal.

The SD is initialized successfully but when you get to the "play" function nothing seems to happen on the output.

I've connected an SD to my Arduino Uno as well as the 1284P and get the exact same problem, pin 9, the output just doesn't output any signal. This is extremely weird and I've checked connections and code 10x over but same problem.

Have you had these kind of problems before, do you know how to locate the problem further or suggest how to fix this?

Thank you very much for your help.

JackTrow, 2015-06-21:
I've got it to work on my Uno, well kind of.

After adding .WAV to the end of the file name it started to play. The extension was not added in the examples.

It plays but it keeps breaking up and the quality is not so good. The output is only around 50mv peak to peak which when connected to a 3.5mm jack into speakers can barley be heard.

TMRh20, 2015-06-21:
Well I heard the same thing from somebody else about adding the file extension, but have never had a problem with that myself, so I have no idea why it would work.

Depending on your sample rate, the library will behave differently. With lower sample rates, you will need to raise the volume. With higher rates, you will want to increase the memory allocated, since the 1284P is being treated like an UNO etc. so is only using minimal memory. (see pcmConfig.h )

Also see http://www.hackerspace-ffm.de/wiki/index.php?title=SimpleSDAudio  (another great audio lib for arduino, though unsure of 1284p support)

closed 2015-10-01 by TMRh20