TMRh20/TMRpcm #26: 3 Audio simultaneously

nurimo, 2015-07-10:
Hi everyone,

I am trying to modfy the code for my application. My goal is playing 2 audio simultaneously using 1 timer and playing the last audio file with another timer. These are done seperately as Multi_Mode and Mode2. I want to combine these 2 features. But I am not a good coder. Can anyone guide me to figure this out ?

TMRh20, 2015-07-13:
Unfortunately, this is not very likely. It was quite a challenge getting a 16mhz arduino to handle two audio tracks simultaneously, and it only handles it well on a Mega, with more memory. 

See https://github.com/TMRh20/TMRpcm/issues/22

nurimo, 2015-07-13:
what if i add some sram on the circuit? I need this to happen. Or can you give me some tips to use Multi_Mode and Mode2 together ?

TMRh20, 2015-07-14:
Its not that simple. The code would have to be re-written entirely, and I'm
not sure it would work at all, except possibly with the lowest of sample
rates. An Arduino Due or Teensy would be much more suited to advanced audio.

nurimo, 2015-07-14:
Sample rate is not important actually for my application. Can you guide me for how to modfy the code or write the code entirely. I cant use Arduino Due due to size problem. I will use this with pro mini or nano. Thanks for all your answers.

mmame, 2015-07-14:
What about using two arduino mini or nano? That's imho the most easy
solution you could get.

nurimo, 2015-07-14:
Yes it was my first idea but application space is to small in every direction.due to this problem i need to figure this out. After i read your post about 2 arduino,it revealed something. But I need to confirm that it can be done. Thanks for it tho.

closed 2015-10-01 by TMRh20