TMRh20/TMRpcm #27: Support for the Due?

MrHaroldA, 2015-07-12:
The Due has the ability to run the PWM ports at 12 bit; which should improve sound quality greatly! It also has two dedicated DAC ports, but they have lots of digital noise and pops when starting a file.

I have two Due's at my disposal, I can help and test if needed.

TMRh20, 2015-07-13:
Since the Due uses completely different architecture etc, I don't have any plans myself to support it with this library.

There is a [basic Due audio library](https://www.arduino.cc/en/Reference/Audio), and there has been some nice work done on the [Teensy audio library](http://www.pjrc.com/teensy/td_libs_Audio.html). Since Teensy and Due share the same processor etc, it may not be too difficult to use on Due, but I am not certain.

MrHaroldA, 2015-07-14:
The Due audio library produces digital noise and pops, and the Teensy Audio System Design Tool lists PWM as inferior to it's 12-bit DACs, and the DAC are listed as noisy ;)

So I guess this is a no-go after all.

closed 2015-07-14 by MrHaroldA