TMRh20/TMRpcm #30: Lower footprint?

RuRios, 2015-09-26:
Just started to play with the library and its working great. However, I don't think I'll be able to use because of its footprint. Any suggestions on how to lower it? Just running a simple sketch (e.g. one of the examples) takes about half of the available storage space on the Arduino I have used.

TMRh20, 2015-09-27:
If you take a look at the default SD card examples in Arduino IDE, you will notice they take nearly half the available storage space, and RAM. 

See https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features and included examples for use with SDFat lib.

The only real solution is to re-write the SD functions for minimalist usage, which has been done in the [SimpleSDAudio](http://www.hackerspace-ffm.de/wiki/index.php?title=SimpleSDAudio) library.

RuRios, 2015-09-27:
Thanks. Will take a look at those. Hopefully with that I can get everything to fit..

RuRios, 2015-09-27:
Unfortunately SDFat does not seem to save much memory and SimpleSDAudio does not allow you to change volume. I guess my only option at the moment is to go with SimpleSDAudio. 

Thanks for your help.

closed 2015-09-27 by RuRios