TMRh20/TMRpcm #31: TMRpcm Freezing Other Processes

Megasonic988, 2015-11-20:
My partner and I are using your TMRpcm library for a school project and we are running into some troubles with it. Before anything, I must say it is a really great piece of work, thank you for making it! The issue we are having is that once a sound file is played, the Arduino freezes, and our following code (graphics rendering on a LCD and its calculations) runs very slowly or not at all until the song ends. We are using an Arduino Mega. I realize that TMRpcm is processor intensive; however, I was wondering if you knew of any possible workarounds or fixes.

TMRh20, 2015-11-21:
Hi, thanks for the compliments!

The library makes heavy use of interrupts to accomplish the task of playing audio separate from the main loop. This makes it relatively easy to use, but it pushes the 16mhz chips to their limits.

The only real thing that can be done is to reduce the audio quality by using a lower sample rate. A sample rate of 11khz is usually not too bad, but nothing like HD audio lol. You can also edit the pcmConfig.h file and change ```#define buffSize 128``` to something larger than 254 on the Mega, to use more memory, but you're not going to see a dramatic improvement.

Your best option if you need to run multiple devices, is probably to look at an external audio module to offload the work.

Megasonic988, 2015-11-26:
Thanks so much for the tips! We ended up going with using a separate audio dedicated to music playback, and things work great now.

closed 2015-11-26 by Megasonic988