TMRh20/TMRpcm #33: Ubuntu 14.04 pcmConfig.h Error

Encom21, 2015-12-31:
Hello there I have a question about an error that I am facing. I download the latest version of TMRpcm and imported to Arduino IDE 1.0.5 for Ubuntu 14.04 and trying to upload to my nano. Then state as following error, (SdAudio.ino :1.68: fatal error: pcmConfig.h: No such file or directory compilation terminated.).  And the weird part is its there can't seem to find a solution so I thought I would report this issue to see if there's a solution.

TMRh20, 2015-12-31:
My initial guess would be that it would be related to the includes. See [here](http://stackoverflow.com/questions/21593/what-is-the-difference-between-include-filename-and-include-filename)

For some reason, I have pcmConfig.h included in both TMRpcm.h and TMRpcm.cpp, so I would try changing the lines ``` #include <pcmConfig.h> ``` to ``` #include "pcmConfig.h" ``` in both files so that it searches the local directory.

Encom21, 2015-12-31:
no it didn't work same error, and i tried to move some of the files around but no success.

TMRh20, 2015-12-31:
It works on a macbook and windows, so it seems to be something platform specific. Try updating IDE or maybe remove the includes for pcmConfig.h, and put any required #defines somewhere else, possibly even renaming the file, but these are just guesses.

gamecompiler, 2017-03-31:
Any progress? Id tryed compiling this lib under MAC, Linux and Windows, everything fine with the current arduino IDE (1.6.13).
What IDE are you using? So i can test out what happend.

Please reopen issue ive theres some new info.

closed 2017-03-31 by gamecompiler