TMRh20/TMRpcm #35: TMRpcm breaks RF24Network

gamecompiler, 2016-01-07:
When i initialize RF24Network with:
    RF24Network network(radio); 

The RF24Network library breaks the TMRpcm library and no sounds play.
But:

    if (!SD.begin(SD_ChipSelectPin)) {  // see if the card is present and can be initialized:
        Serial.println("SD fail");  
        return;   // don't do anything more if not

      }
      else{   
        Serial.println("SD ok");   
      }

Says SD ok.
How can i fix that ? I really would like to use the network library.

Here is my complete code:

    /*
* Giving Credit where Credit Is Due
*
* Portions of this code were derived from code posted in the Arduino forums by Paul Malmsten.
* You can find the original thread here: http://www.arduino.cc/cgi-bin/yabb2/YaBB.pl?num=1176098434
*
* The Audio portion of the code was derived from the Melody tutorial on the Arduino wiki
* You can find the original tutorial here: http://arduino.cc/en/Tutorial/Melody
* 
* The Lasertag portion is taken from the great IBM Lasertag tutorial:
* You can find the original tutorial here:
* http://www.ibm.com/developerworks/opensource/tutorials/os-arduino1/
* 
*/
#include <SD.h>                      // need to include the SD library
//#define SD_ChipSelectPin 53  //example uses hardware SS pin 53 on Mega2560
#define SD_ChipSelectPin 5  //using digital pin 4 on arduino nano 328, can use other pins
#include <TMRpcm.h>           //  also need to include this library...
#include <SPI.h>

//Sound stuff
TMRpcm tmrpcm;   // create an object for use in this sketch
unsigned long time = 0;

//RF24 Stuff
#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

RF24 radio(7,8);                // nRF24L01(+) radio attached using Getting Started board 

//RF24Network network(radio);      // Network uses that radio
const uint16_t this_node = 00;    // Address of our node in Octal format ( 04,031, etc)
const uint16_t other_node = 01;   // Address of the other node in Octal format

struct payload_t {                 // Structure of our payload
  unsigned long ms;
  unsigned long counter;
};
//-------------------------------------------------------
//
//

int sensorPin  = 2;      // Sensor pin 1
int senderPin  = 3;      // Infrared LED on Pin 3
int triggerPin = 4;      // Pushbutton Trigger on Pin 4
int speakerPin = 10;     // Positive Lead on the Piezo
int blinkPin   = 13;     // Positive Leg of the LED we will use to indicate signal is received

int startBit   = 2000;   // This pulse sets the threshold for a transmission start bit
[... 270 lines cut ...]
    // send separation bit
    digitalWrite(senderPin, HIGH);
    delayMicroseconds(waitTime);
  }
  oscillationWrite(senderPin, endBit);
  playTone(100, 5);
  digitalWrite(blinkPin, LOW);
}

void oscillationWrite(int pin, int time) {
  for(int i = 0; i <= time/26; i++) {
    digitalWrite(pin, HIGH);
    delayMicroseconds(13);
    digitalWrite(pin, LOW);
    delayMicroseconds(13);
  }
}

void selfDestruct() {
  myHits  = maxHits+1;
  playTone(1000, 250);
  playTone(750, 250);
  playTone(500, 250);
  playTone(250, 250);
}

TMRh20, 2016-01-09:
It looks to me like you might just need to optimize your code a bit. The SD library uses a pile of memory, so you can 

The first thing I see are memory issues. Lower memory with RF24Network by disabling features:

**Edit RF24Network_config.h** (in the library folder)
1. Uncomment ``` #define DISABLE_FRAGMENTATION  ```
2. Change ``` #define MAIN_BUFFER_SIZE 144 + 10 ``` to  ``` #define MAIN_BUFFER_SIZE 72 + 10 ```
This assumes you are not sending network payloads bigger than 24 bytes, and limits the number of cached payloads to 3 at the RF24Network layer, and 3 in the radio buffers.

**Use Flash/Program Memory for Strings**
1. Instead of Serial.print("This string"); just use Serial.print(F("This string"));
2. You can also store wav file names in flash. See https://github.com/TMRh20/TMRpcm/blob/master/examples/progmem/progmem.ino 

This will store all of your strings in program memory, saving precious SRAM for other things.

You can also modify and adjust settings in the pcmConfig.h file, specifically using the SDFat library may lower memory usage a bit too.

gamecompiler, 2017-01-02:
Thank you for your help.
With the help of your awnsers i solved it!
Thank you very much 😄

closed 2017-01-02 by gamecompiler