TMRh20/TMRpcm #36: Slow Audio Playback

RobertDowns, 2016-01-13:
I have tried using TMRpcm with a Catalex micro sd reader. I converted the wav to a couple of different sample frequencies and they have been done at 8 bit, I can get the sound file to play, but it does so slowly, so the speech is slowed down by at least a factor of two or more?

Any ideas on what I've done wrong?

cheers
Rob

TMRh20, 2016-01-13:
They need to be both 8-bit and mono. Sample rates between 8-32khz should be picked up automatically, so that is generally not an issue. I would guess that you have 'double samples' or something, as you would with 16-bit or stereo audio.

The [wiki](https://github.com/TMRh20/TMRpcm/wiki) has a quick step by step for audacity or iTunes that should work.

RobertDowns, 2016-01-13:
Stereo audio it was. Working perfectly now.

Cheers

closed 2017-05-23 by gamecompiler

hosjiu1702, 2018-04-25:
I will explain more @RobertDowns's idea.
In pcmConfig.h file uncomment #define STEREO_OR_16BIT