TMRh20/TMRpcm #39: Buzzing Sound While Using TMRpcm

BlueBear365, 2016-02-08:
Hi,

I am a newbie and doing a school project to learn more on Arduino.

I have Arduino Uno and ethernet shield. I have connected these two and also have put a formatted micro SD card in the SD card reader of the ethernet shield. I have connected a basic speaker from a local hobby shop on pin 9 and ground.

When I try to run a program at http://www.arduino.cc/en/Tutorial/Tone, it works perfectly fine. The only minor change is that I have used pin 9 instead of pin 8, and this tutorial works perfectly fine for me.

My next goal is to play a simple audio file.

For this, I am trying to use TMRpcm, the same circuit and sample programs giving me buzzing sound on my output. Here is my code, but I believe the issue may be with my circuit itself:

#include <SD.h>                      
#define SD_ChipSelectPin 4  
#include <TMRpcm.h>        

TMRpcm tmrpcm;   
char mychar;

void setup(){

  tmrpcm.speakerPin = 9; 

  Serial.begin(9600);
  if (!SD.begin(SD_ChipSelectPin)) {  // see if the card is present and can be initialized:
    Serial.println("SD fail");  
    return;   // don't do anything more if not
    Serial.println("SD successful");  
  }
  tmrpcm.setVolume(7);
  tmrpcm.quality(1);

  tmrpcm.play("DRUM1.WAV");   // sample file I created using given specification using audacity

}

void loop() {
  Serial.print("In loop");  // This is just a dummy code
  delay(5000);

}

This just brings up some buzzing sound for about 20 seconds. Can someone help with point me to right circuit for this, or point me to a simple audio file player?

Thanks.

erdinckutuk, 2016-02-09:
hi..
I guess you are using nano. 
Volume adjustment for nano should be between 5 and 2 for best performance.

BlueBear365, 2016-02-09:
I tried volume levels 2,3 and 5. There is no change.

Another thing to notice in the issue is, the buzzing sound (20-25 seconds) is much longer than the original wav file length (2-3 seconds). I do not know what is going on.

amoln9, 2016-05-11:
@BlueBear365 
m also facing same problem. Did u get the answer

themarquee, 2016-07-05:
Same issue using Uno, Ethernet shield, FAT16 formatted SD, and Audacity exported WAVs (unsigned 8 bit, Microsoft wav, 16kHz).

Tones work fine, but following the examples for wavs gives me either buzzing or silence.

Anyone figured this issue out? Is there a test file or sound I can use to rule out any conversion/compatibility issues?

amoln9, 2016-08-08:
it is the problem associated with wave files. try downloading some sample wav files (8 bit 16Khz).
i have downloaded some audio files of alphabets and I'm able to play them very well(with some dc noise and low pass filter).

chaosbiber, 2018-03-28:
It's an old issue, but I had the same problem. As in the first post the **Ethernet shield** is mentioned, this should be the solution:
http://forum.arduino.cc/index.php?topic=313974.0
In pcmConfig.h uncomment
``//#define DISABLE_SPEAKER2``

tomperys, 2019-04-20:
> It's an old issue, but I had the same problem. As in the first post the **Ethernet shield** is mentioned, this should be the solution:
> http://forum.arduino.cc/index.php?topic=313974.0
> In pcmConfig.h uncomment
> `//#define DISABLE_SPEAKER2`

Thanks, I have problem with buzzing sound using the ethernet shield and it took me several hours to find this thread and the solution worked!!!!

closed 2020-05-11 by TMRh20