TMRh20/TMRpcm #40: Goes quiet after repeated plays

bowlercaptain, 2016-02-11:
Hello! I'm having another issue. I'm using the same setup as I mentioned in my [previous issue](https://github.com/TMRh20/TMRpcm/issues/38), (slightly sketchy arduino leonardo and SD card reader), and am successfully playing great sound. However, After a number of plays (seemingly random - sometimes as high as fifty or a hundred and once as low as two (after increasing buffer size)), the sound will cut out, and nothing more will play. This happens in both multi and single mode, MODE2 or not. Our original code responded to inputs using input_pullup and force-sensitive resistors connected to ground, but we get the same effect with just:

loop(){
tmrpcm.play("hat.wav",0);
delay(1000);
}

We are playing a high number of short sound files, and are wondering if there could be a problem with random repeated SD card reads or a memory issue playing multiple files (or even one file played multiple times). Any suggestions?

erdinckutuk, 2016-02-22:
Can you try SDFat.
SD library have a bug for repeating read.

bowlercaptain, 2016-03-14:
Issue turned out to be memory-based. What I didn't mention (because I didn't notice it) was the loud click after the sound played. The system went silent after fewer repeats if the buffer size was too large and would also fail if the buffer size was too small. After switching from multi to single and playing around with our buffer size (the sweet spot was 230) the clicking stopped and the project worked.

closed 2016-03-14 by bowlercaptain