TMRh20/TMRpcm #41: im not able to compile

ashkid, 2016-03-04:
Arduino: 1.6.7 (Windows 8.1), Board: "Arduino/Genuino Uno"

C:\Users\Diksha\AppData\Local\Temp\arduino_5484f871416f9ce1317bba18f537edf5\basic.ino: In function 'void setup()':

C:\Users\Diksha\AppData\Local\Temp\arduino_5484f871416f9ce1317bba18f537edf5\basic.ino:19:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   tmrpcm.play("music"); //the sound file "music" will play each time the arduino powers up, or is reset

                      ^

C:\Users\Diksha\AppData\Local\Temp\arduino_5484f871416f9ce1317bba18f537edf5\basic.ino: In function 'void loop()':

C:\Users\Diksha\AppData\Local\Temp\arduino_5484f871416f9ce1317bba18f537edf5\basic.ino:28:26: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       tmrpcm.play("music");

                          ^

C:\Users\Diksha\Documents\Arduino\libraries\arduino_420772\pcmRF.cpp:8:20: fatal error: SdFat.h: No such file or directory

  #include <SdFat.h>

                    ^

compilation terminated.

Multiple libraries were found for "SD.h"
 Used: C:\Users\Diksha\Documents\Arduino\libraries\SD
 Not used: C:\Users\Diksha\Documents\important\softwares\Arduino\libraries\SD
exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

lnvona, 2017-04-19:
Was this issue solved??? I have the same problem.

gamecompiler, 2017-04-20:
Using a fresh copy of the current Arduino IDE it works perfectly.

---------------------------------------------
Der perfekte Zeitpunkt seinen Stromanbieter zu wechseln:
http://www.anpa.de/index.php?action=stromvergleich

closed 2017-05-23 by gamecompiler