TMRh20/TMRpcm #42: Explanation of 'complimentary mode'

gdhgdhgdh, 2016-03-25:
Hullo :)

Can you explain what the 'complimentary mode' is for? I see it referenced in the Advanced Usage page and on the wiki but can't see its actual purpose.

Cheers,
Gavin.

TMRh20, 2016-03-25:
2 pins (speaker) = push / push (not complimentary)

2 pins (speaker) = push / pull (complimentary)

gdhgdhgdh, 2016-03-26:
Thanks for that - I still don't really get it. I understand that better volume from the speaker will be achieved by having outputting both positive and negative voltages so the cone travels further.

In Push/Pull mode, are two output pins being manipulated to deliver this setup (instead of a typical single pin and GND) ?

TMRh20, 2016-03-27:
That is exactly it, one speaker pin to ground == push/null & with 2 pins
each pin can be positive or negative so is push/push.

TMRh20, 2016-03-27:
oops, meant push/pull instead of push push for clarity.

On 2016-03-26, at 7:55 PM, TMRh20Projects <tmrh20@gmail.com> wrote:

That is exactly it, one speaker pin to ground == push/null & with 2 pins
each pin can be positive or negative so is push/push.

closed 2020-05-11 by TMRh20