TMRh20/TMRpcm #44: play song one after another in a loop

simon11111, 2016-04-06:
hi master

i have tried to put wav.play("music1.wav"); and wav.play("music2.wav"); in the setup()
it worked
but when I placed these files into the loop
it cannot play both files   it only played one 

is there a way to play my music one after another without pressing any button?

Thanks a lot.

TMRh20, 2016-04-06:
Something like this? See the progmem example maybe too

```
#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>

#define SD_ChipSelectPin 4

TMRpcm wav;

char *wavs[] = {"music1.wav", "music2.wav", "music3.wav"};
uint8_t counter;

void setup() {
  wav.speakerPin = 9;
  Serial.begin(9600);
  SD.begin(SD_ChipSelectPin);
}

void loop() {
  if (!wav.isPlaying()) {
    wav.play(wavs[counter]);
    counter++;
    ++counter %= 3;
  }
}
```

simon11111, 2016-04-07:
Thanks a lot! It worked! :)

Kilsan7, 2017-05-21:
You're genius!!!

gamecompiler, 2017-05-23:
Seems solved :)

closed 2017-05-23 by gamecompiler