TMRh20/TMRpcm #45: PWM signal after music stops

simonvu14, 2016-05-17:
Hi there,
I would like to report a bug, or it is not really a bug. I am not sure.

I tried this on Arduino Uno, Leonardo and mega328 alone, and it is very consistent.

Setup oscilloscope on output D9 I got this
[image]
32kHz Square wave about 25% duty circle.

Connect D9 to RC lowpass filter 1k Ohm and 15nF which has cut-off frequency about 10kHz
[image]

You can see the amplitude of 32Khz is still  quite high.

I got sawtooth signal on the output
[image]

Of course I can change the cap to 68nF which cut-off frequency is about 2kHz to eliminate 32Khz square wave, but it also cut-off part of music too.

It happens right after playing the first song. And it stays there even after music stops. It is quite annoying. With that noise like that, I am not sure connect it to professional audio equipment is safe, even with a coupling capacitor.

Like I said, I am not sure this is a bug or it happens because of the mechanism of generate analog signal from PWM. 

Best regards,
Simon Vu

TMRh20, 2016-05-18:
The audio is reproduced using PWM & timers, and is left on between tracks to keep things simple when playing multiple sounds etc. To disable, see the [ ``` audio.disable() ``` ](https://github.com/TMRh20/TMRpcm/wiki#functions) function.

simonvu14, 2016-05-18:
It is how PWM and timers work, so it is inevitable. 
Then I have to work on lowpass filter to filter it out

Thanks for fast reply,
Simon Vu

joneser005, 2018-01-12:
Probably related to this issue: After wav playback is complete, a signal ~2v stays on the line.  It doesn't produce sound - I only noticed because I had my project (a [race tree](https://www.instructables.com/id/Radio-Controlled-Race-Tree/)) hooked into a high-dollar sound board and the sound guy was a little concerned about that.  I later discovered calling audio.disable() after playback eliminates the stray voltage.  I only mention that here in case this behavior is a bug.  For all I know, you intended it to be this way for some reason.

TMRh20 - thank you for your excellent work on this library!

TMRh20, 2018-01-13:
This is intended. I leave the timers running and ramp the signal up or down to prevent popping and clicking between audio tracks.

closed 2020-05-11 by TMRh20