TMRh20/TMRpcm #46: Continuous popping noise when Arduino powered via VIN

alexissantos, 2016-06-05:
When I supply power (12V or 5V) to the Arduino (an Adafruit Metro Mini) via VIN, a continuous popping sound occurs when audio is supposed to be playing, and the light on the SD card reader and the pin 13 light on the Arduino flashes along with the pop. When power is supplied over USB, audio playback works fine.

Here's a short video of the pop: https://www.youtube.com/watch?v=3xnE4kFzT7E

Any idea why this might be the case? Maybe it's a fault of [my amp](https://www.adafruit.com/product/987) and not this library?

TMRh20, 2016-06-05:
SD cards can draw up to 200mA . The regulator is only 150mA.

alexissantos, 2016-06-05:
Aha! Just to make sure I've got this right: I've got a 2A power supply, which should be enough, but the voltage regulator on the board connected to the VIN pin is only making 150mA of it usable?

closed 2016-06-05 by alexissantos

TMRh20, 2016-06-05:
Yeah, I think so

alexissantos, 2016-06-05:
Thanks! I'll pick up a 5V regulator to power the SD card reader.