TMRh20/TMRpcm #49: ESP8266 port

gamecompiler, 2016-08-06:
I really would like to use your awesome nice library on a esp8266.
kirchnet@ the esp8266 community wrote a "good" working code to play a wav file:
http://www.esp8266.com/viewtopic.php?p=44557
(Sound a lot of more awfull than your great library on a ATmega(Compared to his code yours sounds like HiFi))

The code still has some problems, but anyway,
is a esp8266 port planned ?

Sorry for my english.
Thanks in advance,
gamecompiler.

TMRh20, 2016-08-07:
Well, IMHO AVR & Arduino devices are a lot of fun to play with, where with
ESP devices, I find myself struggling to find proper documentation and
examples. If anyone can point me to some proper documentation regarding
internal registers & timers etc,(like a proper datasheet) that would be
awesome.

gamecompiler, 2016-08-16:
I can give you that files that i got:

//Hardware stuff

https://espressif.com/sites/default/files/documentation/0b-esp8266_system_description_en.pdf
https://espressif.com/sites/default/files/documentation/esp8266-hardware_matching_guide_en.pdf
https://espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf
https://espressif.com/sites/default/files/documentation/esp-wroom-s2_datasheet_en.pdf
https://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf

//API Stuff

https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf
https://espressif.com/sites/default/files/documentatio/20b-esp8266_rtos_sdk_api_reference_v1.4.0_0.pdf

Thats the actual stuff ive found. (Documents got updated some days ago, the links are the "fresh" ones)

fredericplante, 2016-10-30:
http://arduino.esp8266.com/stable/package_esp8266com_index.json

In your Arduino config file. It will appear in your board manager. Install it from there. Man your library would totally be great on the esp8266. Also there is a flash drive on the beast so for simple sound, it would totally fit.

And even for recording, there is an ADC 0v to 1V on board, so the CB aspect of your project could be doable.

I will say Hurray! if you ever make a fix for ESP8266. :+1: :D

codahq, 2016-11-20:
One more vote for you to add support.

padlnpack, 2016-11-24:
I have a project I need to play wav files and it is cost sensitive. I hope the library can be ported for 8266.

fredericplante, 2016-11-24:
To me, the best way to go would be to program the esp8266 as a SPI  client that emulate the NRF24 module but TCPIP wize, then connect it to an plain arduino. Basically use the library like it is and morph the ESP8266 to make the arduino believe that it just a NRF24.

gamecompiler, 2016-11-24:
@fredericplante I think that this has nothing to do with that issue. What every you mean, please open a own issue for that. Its cool guys that you all want the library to be ported. But nobody helps anybody here when saying "That would be awesome". Please stop spamming. @TMRh20 and I have a full time job. So please sit back and relax. Im on it to get it work. But this needs time. You guys maybe know what i mean.

fredericplante, 2016-11-24:
Wow you got an amazing ego trip, maybe it's time you go take you pills :D @gamecompiler gamecompiler Why don't you do it, if you feel you are so amazing.

TMRh20, 2016-11-24:
Ok everybody out of the pool.

Time to take a break folks.

msurguy, 2017-01-25:
Just wanted to say thank you to the library maintainers, this lib has impacted a lot of projects in a very positive way. Keep it up!

gamecompiler, 2017-01-29:
Sorry for the long delay. I study ET/IT and i write a lot of exams this semester.
Will need a little bit longer than expected.
I currently do a complete rewrite of the lib. Also trying to implement a "streaming feature".
Got it almost work using RTP.  (RIFF wav stream)
Using the esp8266s timer is also a big problem because nobody really knows if the timers are used....
For example if using wlan timer0 is automatically used. Currently im using a software timer to decode wav files. The sound quality is like expected very bad....!
Also a lot of problems i ran are overflows, if not using yield().

The esp8266 is a very cool device - but the documentation of the arduino IDE addon and of the device itself is just awfull.

gamecompiler, 2017-05-23:
Argh, got so many crashes. Will need more time because of my many exams...

msurguy, 2017-06-19:
@gamecompiler have you tried using PlatformIO? I think it might make the further development for many boards a whole lot easier in terms of compiling / testing on multiple devices at once

gamecompiler, 2017-08-16:
Im sorry, but id stoped the development. Im using an raspberry pi zero now for my project.
Id ran in too much problems using the timer. While wifi enabled nothing worked etc.

Im really sorry.

closed 2017-08-16 by gamecompiler

codahq, 2017-08-17:
Bummer.

gamecompiler, 2017-08-17:
@codahq Why didnt you made it yourself ;)

codahq, 2017-08-17:
Who knows.  Maybe I will get desperate enough that I will.  I have no availability right now for a project like this though.