TMRh20/TMRpcm #50: Example Not Working for Mega 2560

MattTucker22689, 2016-08-25:
I have tried multiple times to get the example code "music" to work, with no luck. I am using an Arduino Mega 2560 SainSmart clone, and a Seeed V4.1 SD Shield. I have tried multiple coding and wiring configurations.

I HAVE been able to get the CardInfo example to work. So I know(or at least am almost certain) the neither board is damaged in any way. I even went so far as to try a Catalex Micro SD board(the one I do suspect to be ruined- none of wiring configurations I tried worked with CardInfo)

It is my goal, to eventually get my project to both move and produce sound, and am trying to tackle each task individually and then combine them.

Help would be MUCH APPRECIATED!!!!

TMRh20, 2016-08-29:
Power is often an issue with SD cards/modules. This is not an issue with
many standard sketches, because they are not continuously reading, thus not
using much power. I would suggest a separate power supply for the SD card,
and make sure to connect GND to Arduino GND.

Movement via servos could be done simply by using a timer. I've thought
about making a servo library compatible with the audio lib, but haven't
gotten around to it.

peace,
TMRh20

gamecompiler, 2017-03-30:
Please post your code here, otherwise we cant help you MattTucker22689.
I close this issue, because we didnt got any answer from you a long time.
You can always reopen it! Just to make sure.

closed 2017-03-30 by gamecompiler