TMRh20/TMRpcm #51: Support for Arduino 101?

rdpoor, 2016-10-02:
Any plans to support the Intel-based Arduino 101?

TMRh20, 2016-10-08:
Yup, it probably will work as-is or require only a few changes. 

For anybody that has one, feel free to post issues.

TMRh20, 2016-10-08:
Wait a second, I thought this was regarding the RF24 library. In this case, probably not anytime soon.

rdpoor, 2016-10-13:
FWIW, I wrote a PCM-based audio DAC for the Arduino 101 using ARC_V2_TMR1 for the sample clock and one of the Quark PWM timers for PWM.  Was really rather simple.  We couldn't use it in our production system since the servo motors also require TMR1 and I haven't re-written it to use just Quark timers, but that appears relatively straightforward.

cebaker, 2018-04-02:
Hey I am working on a similar idea but do not need the servo motors. Would you happen to still have the work around or examples of how to do it in you public repositories.  Thanks

rdpoor, 2018-04-03:
Let me dig through my archives -- no promises!  :)

rdpoor, 2018-04-03:
Hi @cebaker  -- no promises, but try this:

    https://github.com/rdpoor/arduino101-pwm-dac

Let me know if it works for you or if you get stuck.

If the former, consider "hardening" the code and creating a pull request for TMRpcm.  If the latter, let me know and I'll dust off my Intel Curie board to see if I can reproduce the problem.

cebaker, 2018-04-03:
Well after days of work on this I must admit I have given up. I was trying to record 2 seconds of audio and save it into program memory then run this wav file in our already trained neural network for voice recognition. So I also needed to figure out a way to record into program mem. This and some more libraries where needed that where on the Arduino Uno . Because of this I emailed  my professor just a few hours ago and he informed me this morning that he might have an Arduino Uno with sd card shield which you may know supports the TMRpcm library and others important libraries for recording .I will meet with him today in his office but if he doesn’t have it I will put more work down this path.

P.s. Sorry for the long response , just wanted to fully explain.  I will link my repository with the full project after I am done if you are interested.

https://github.com/rdpoor/arduino101-pwm-dac

closed 2020-05-11 by TMRh20