TMRh20/TMRpcm #52: Starting Music and Pwm Signal Down

fearles577, 2016-10-07:
Hi, 

im control arduino with RC car transmitter. receiver signal pins connected A0, A1, A2.
İm read signal 1000-2000 and mapping -255/255. 0 (Zero) center position. 

Speaker connected 9. Pin on Arduino Uno

İm starting arduino, transmitter at center position and serial print analog pins 0 value. 
İm doing button up Serial print 255 and button down Serial print -255.
No problem so far...
Again transmitter center position, Serial Print 
A0: 0
A1: 0
A2: 0

only start playing wav and later 

A0: -255
A1: -255
A2: -255

Music continues playing, Button down
A0: -255
A1: -255
A2: -255

Music continues playing, Button up

A0: -130
A1: -130
A2: -130

And stop play music.
returns to normal
A0: 0
A1: 0
A2: 0

I would be very happy if you help

Thanks

TMRh20, 2016-10-08:
I'm not sure I understand the problem? Can you post your code, along with a more detailed explanation of what is happening?

gamecompiler, 2017-03-30:
Im sure he didnt even read your answer TMRh20. fearless577, please reopen this issue and post your code here if you read this.

closed 2017-03-30 by gamecompiler