TMRh20/TMRpcm #56: Recording not working with SdFat

SilverGreen93, 2016-12-30:
I tried to use the recording example, but if I
#define SDFAT
#define ENABLE_RECORDING
#define BLOCK_COUNT 10000UL 
The example does not compile because SDVolume is deprecated but it is used by  TMRpcm::createWavTemplate
How can I replace SdVolume with something that works?

I cannot use the default SD.h. It does not work for me even in the basic examples. Only SdFat works.

gamecompiler, 2017-02-12:
Please give us your complete arduino error log.
Thanks in advance.

SilverGreen93, 2017-02-15:
I used the SdFat library from here: https://github.com/greiman/SdFat
and if I try to compile I get the error that SdVolume is not declared.
Meanwhile I found another version (maybe older version?) of SdFat and it works fine with it.
I will come back with the full error output if it is needed anymore, but for now it works with the other version that I have downloaded.

gamecompiler, 2017-02-15:
Ok, i close this issue until you reopen it.

closed 2017-02-15 by gamecompiler