TMRh20/TMRpcm #57: Possible to set different PWM frequency for playback

Comiter, 2017-01-06:
Hi,
do anyone know if it possible to alter the PWM frequency on the Audio Out pin?
I would be happy to exchange resolution for increased PWM frequency (to be able to remove the switching noise from the playback).

gamecompiler, 2017-02-19:
Id got no switching noise, but that wouldnt be the correct solution to fix "switching noise".
Could you please record you audio? Than i may be able to help you.

If we change the PWM frequency that wouldnt have any effect.
Only if we change the overall sampling rate - INCLUDING the PWM frequency.
(Please correct me if im wrong) 

What you can do to increase audio quality is to build a low pass filter. Just google for it and you audio will be 100% better (Not hifi, but its quite nice :D)

gamecompiler, 2017-03-30:
Did you try the low pass filter?
Please reopen issue if you need help.

closed 2017-03-30 by gamecompiler