TMRh20/TMRpcm #58: TMRpcm has no member named ‘startRecording’

MrLeonardo, 2017-01-21:
I have this error:

error: ‘class TMRpcm’ has no member named ‘startRecording’.

Why?

gamecompiler, 2017-02-12:
First, please show us your complete arduino error log. (The lib works for me and others) Otherwise we cant help you.
And before that, please check if you got the newest version of TMRpcm library.

Thanks in advance.

rodrigo-martins, 2017-02-19:
The same issue for me. Ive got the 'recording' exemple

gamecompiler, 2017-02-19:
When people got that problem, and i not! Why nobody cant show his error log? Whats wrong with you guys? We need an error log, not the information "Its not working". I close this issue now, can be reopened if somebody post his FULL error log. Im sorry.

rodrigo-martins, 2017-02-19:
C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino: In function 'void loop()':

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino:55:59: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 'r': audio.startRecording("test.wav",16000,A0); break;    //Record at 16khz sample rate on pin A0

                                                           ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino:56:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 'R': audio.startRecording("test.wav",16000,A0,1); break;  //Record, but with passthrough to speaker.

                                                             ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino:57:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 't': audio.startRecording("test.wav",16000,A0,2); break;  //Do not record. Output direct to speaker

                                                             ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino:60:49: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 's': audio.stopRecording("test.wav"); break;              //Stop recording

                                                 ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino:61:40: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 'p': audio.play("test.wav"); break;                       //Play the recording 

                                        ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp: In member function 'byte TMRpcm::metaInfo(boolean, char*, char*, byte)':

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1381:18: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   char* datStr = "LIST";

                  ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1382:28: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   if(infoType == 1){datStr = "ID3 "; datStr[3] = 3;}

                            ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1410:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

  char* tagNames[] = {"INAM","IART","IPRD"};

                                          ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1410:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1410:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1430:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    tagNames[0] = "TPE1"; tagNames[1] ="TIT2"; tagNames[2] ="TALB";

                ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1430:38: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    tagNames[0] = "TPE1"; tagNames[1] ="TIT2"; tagNames[2] ="TALB";

                                      ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1430:59: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    tagNames[0] = "TPE1"; tagNames[1] ="TIT2"; tagNames[2] ="TALB";

                                                           ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp: In member function 'void TMRpcm::finalizeWavTemplate(char*)':

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1568:65: warning: narrowing conversion of '(fSize >> 16)' from 'long unsigned int' to 'byte {aka unsigned char}' inside { } [-Wnarrowing]

  seek(4); byte data[4] = {lowByte(fSize),highByte(fSize), fSize >> 16,fSize >> 24};

                                                                 ^

C:\Users\Pinipa\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1568:77: warning: narrowing conversion of '(fSize >> 24)' from 'long unsigned int' to 'byte {aka unsigned char}' inside { } [-Wnarrowing]

  seek(4); byte data[4] = {lowByte(fSize),highByte(fSize), fSize >> 16,fSize >> 24};

                                                                             ^

gamecompiler, 2017-02-19:
Thank you very much for your error log, i will take a look at it next weekend.

MrLeonardo, 2017-02-20:
Excuse my delay 

there is the log in Windows 10:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Giovanni\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\Giovanni\AppData\Local\Temp\arduino_build_748663 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Giovanni\Downloads\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Giovanni\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\Giovanni\AppData\Local\Temp\arduino_build_748663 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Giovanni\Downloads\TMRpcm-master\examples\xtraFeatures\Recording\Recording.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Opzioni di compilazione cambiate, ricompilo tutto
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\Giovanni\AppData\Local\Temp\arduino_build_748663\sketch\Recording.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\SD\src" "C:\Users\Giovanni\AppData\Local\Temp\arduino_build_748663\sketch\Recording.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\SD\src" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\Giovanni\AppData\Local\Temp\arduino_build_748663\sketch\Recording.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\SD\src" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\Giovanni\Documents\Arduino\libraries\TMRpcm" "C:\Users\Giovanni\AppData\Local\Temp\arduino_build_748663\sketch\Recording.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\SD\src" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\Giovanni\Documents\Arduino\libraries\TMRpcm" "C:\Program Files (x86)\Arduino\libraries\SD\src\File.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DAR [... cut ...]
         case 'p': audio.play("test.wav"); break;                       //Play the recording 

                                        ^

Uso la libreria SD alla versione 1.1.1 nella cartella: C:\Program Files (x86)\Arduino\libraries\SD 
Uso la libreria SPI alla versione 1.0 nella cartella: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Uso la libreria TMRpcm alla versione 1.0 nella cartella: C:\Users\Giovanni\Documents\Arduino\libraries\TMRpcm 
exit status 1
'class TMRpcm' has no member named 'startRecording'

TMRh20, 2017-02-22:
See the wiki and pcmConfig.h to enable recording etc

MrLeonardo, 2017-02-27:
Thanks :-)

Leonardo Capone

closed 2017-03-02 by gamecompiler

amine179, 2019-02-27:
please i have the same error as MrLeonardo, how this was solved i cant ??

MrLeonardo, 2019-02-27:
Hi,
check the library that is included in the project because I had a previous
version.

Leonardo

HaoJuUESTC, 2019-03-01:
I already modified the pcmConfig.h as was told, but when running the "Recording" template, I still receive the same error...
Mine went like this:

D:\Software\arduino-1.8.7-windows\arduino-1.8.7\arduino-builder -dump-prefs -logger=machine -hardware D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware -tools D:\Software\arduino-1.8.7-windows\arduino-1.8.7\tools-builder -tools D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -built-in-libraries D:\Software\arduino-1.8.7-windows\arduino-1.8.7\libraries -libraries C:\Users\Hao Ju\Documents\Arduino\libraries -fqbn=arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10807 -build-path C:\Users\HAOJU~1\AppData\Local\Temp\arduino_build_576162 -warnings=none -build-cache C:\Users\HAOJU~1\AppData\Local\Temp\arduino_cache_857312 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avrdude.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -verbose C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\examples\xtraFeatures\Recording\Recording.ino
D:\Software\arduino-1.8.7-windows\arduino-1.8.7\arduino-builder -compile -logger=machine -hardware D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware -tools D:\Software\arduino-1.8.7-windows\arduino-1.8.7\tools-builder -tools D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -built-in-libraries D:\Software\arduino-1.8.7-windows\arduino-1.8.7\libraries -libraries C:\Users\Hao Ju\Documents\Arduino\libraries -fqbn=arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10807 -build-path C:\Users\HAOJU~1\AppData\Local\Temp\arduino_build_576162 -warnings=none -build-cache C:\Users\HAOJU~1\AppData\Local\Temp\arduino_cache_857312 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avrdude.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\tools\avr -verbose C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\examples\xtraFeatures\Recording\Recording.ino
Using board 'pro' from platform in folder: D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\arduino\avr
Using core 'arduino' from platform in folder: D:\Software\arduino-1.8.7-windows\arduino-1.8.7\hardware\arduino\avr
Detecting libraries used...
"D:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\cores\\arduino" "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\HAOJU~1\\AppData\\Local\\Temp\\arduino_build_576162\\sketch\\Recording.ino.cpp" -o nul
"D:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\cores\\arduino" "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\libraries\\SD\\src" "C:\\Users\\HAOJU~1\\AppData\\Local\\Temp\\arduino_build_576162\\sketch\\Recording.ino.cpp" -o nul
"D:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\cores\\arduino" "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\libraries\\SD\\src" "-ID:\\Software\\arduino-1.8.7-windows\\arduino-1.8.7\\hardware\\arduino\\avr\\libraries\\SPI\\src" "C:\\Users\\HAOJU~1\\AppData\\Local\\Temp\\arduino_build_576162\\sketch\\Recording.ino.cpp" -o nul
[... cut ...]
                         ^

Recording:57:25: error: 'class TMRpcm' has no member named 'startRecording'

         case 't': audio.startRecording("test.wav",16000,A0,2); break;  //Do not record. Output direct to speaker

                         ^

Recording:60:25: error: 'class TMRpcm' has no member named 'stopRecording'

         case 's': audio.stopRecording("test.wav"); break;              //Stop recording

                         ^

exit status 1
'class TMRpcm' has no member named 'startRecording'

MrLeonardo, 2019-03-01:
Where is TMRpcm library? Download the project from github and delete all
other version in your pc.

Leonardo

zzmbird, 2019-12-05:
I already modified the pcmConfig.h as was told, but when running the "Recording" template, I still receive the same error...

zzmbird, 2019-12-09:
@TMRh20 I already modified the pcmConfig.h as was told, but when running , I still receive the same error...What else can i do?plz.

here are my code:

#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>
#define SD_ChipSelectPin 10
TMRpcm audio;
void setup() {
audio.speakerPin = 4;
Serial.begin(115200);
if (!SD.begin(SD_ChipSelectPin)) {
Serial.println("SD Fail");
return;
}else{
Serial.println("SD OK");
}
audio.CSPin = SD_ChipSelectPin;
}
void loop() {
if(Serial.available()){
char c = Serial.read();
Serial.println(c);
switch(c){
case 'r': audio.startRecording("test.wav",16000,A0); break;
case 's': audio.stopRecording("test.wav"); break;
}
}
}

i got those errors:

Arduino:1.8.10 (Mac OS X), 开发板："Arduino/Genuino Uno"

/Users/apple/Documents/Arduino/sketch_dec07b/sketch_dec07b.ino: In function 'void loop()':
sketch_dec07b:30:25: error: 'class TMRpcm' has no member named 'startRecording'
case 'r': audio.startRecording("test.wav",16000,A0); break;
^~~~~~~~~~~~~~
sketch_dec07b:31:25: error: 'class TMRpcm' has no member named 'stopRecording'
case 's': audio.stopRecording("test.wav"); break;
^~~~~~~~~~~~~
'class TMRpcm' has no member named 'startRecording'

TMRh20, 2019-12-14:
First things first, please don’t spam multiple issues looking for an answer. Not helpful.

Second, you most likely have multiple versions of TMRpcm installed. Try deleting the library using library manager, then open up your library folder and delete it there as well if it exists. Then reinstall using library manager. It should work after that.

semajoel, 2020-08-28:
Thanks all, changes in pcmConfig.h file solved this error for me