TMRh20/TMRpcm #60: Mix two signals into one output pin

ChrissHeIs, 2017-02-28:
Hi. Thanks for a great lib! I have a small question.

I have a project that implies one speaker(connected via a single transistor amplifier) and a possibility to play background music with some occasional sounds over it. I was wondering if it is possible to mix two files playing to a single output pin in ENABLE_MULTI mode? After several hours of exploring the code I was able to pause the file1 while file2 is being played (I've modified `ISR(TIMER2_OVF_vect)`, buffer2 is directed to *OCRnA[tt] and `if(playing)` condition at s1078 became `if(playing && !playing2)`) but I still wasn't able to actually mix buffer and buffer2 together. Do you have any ideas how I could do that? If not maybe you know some way to mix the two outputs schematically? (however the programatic approach is preferable as it frees an output pin, which are quite limited for my project)

Thanks in advance.

gamecompiler, 2017-03-02:
Thats i really good question Oo!
Im not an sound expert, do you have a source how sound mixing goes in "math" ?
I dont know how its possible. Im sorry, may TMRh20 knows something about that.
If id understood everything correctly, you want to mathematically mix your two buffers into one and play it. 

"Sound mixing" would be a very cool feature!
Thank you very much for the idea.

With king regards,
gamecompiler

GolPesar132, 2017-10-03:
I think bigger value between two songs must play at each time.
Can you test it?

TJWi, 2018-05-27:
I found this research paper during my search for alternate solution, hope it might help make this enhancement move forward.

https://www.researchgate.net/publication/275656940_Digital_sound_processing_using_arduino_and_MATLAB

closed 2020-05-11 by TMRh20