TMRh20/TMRpcm #61: ATmega32U4 with 16 bit resolution

frank8051, 2017-03-01:
Hi,
I am trying to get a "Pro Micro" board (ATmega32U4) running with 16 bit mono. While everything works fine with 8 bit, enabling 16 bit produced stange output. I found out that `__AVR_ATmega32U4__` seems not to be defined for the Pro Micro board, so I took "Arduino Leonardo" in the IDE. Now the output is shown here:

[image]

This is the result of the upper 8 bits of a 1kHz sine wave, sampled with 32 kHz. On the left side it looks like the already fixed sign bug, on the right side there are periodical peaks. The whole sequence is repeating.

The sample was created with Audacity and stored as Microsoft WAV, I also tried `sox` for generating samples. 8 bit resolution with stereo seems to work.

Any ideas?

Thanks in advance!
BR Frank

mmame, 2017-03-02:
Hi Frank

First of all, 16 bit mono with 32kHz may be too much running on a ATmega32U4.
Try lowering the sample rate.
Unfortunately, the TMRpcm library still has issues preventing with 16 bit
sound. These are:

- Imho, rampUp uses only one byte even in 16 bit mode. This results in an
"odd" index which will push the wrong 16 bit "pair" on the pwm output (it
mixes up the lower 8 bits of two 16 bit values). I didn't had time to push
a fix for this so far. As a simple workaround, you can increase the Index
on RampUp by one to re-align the index.

- I was never able to get "HIFI" sound with 2R resistor ladder. Thus I
decided to use a SPI DAC instead. The SPI can either be implemented with
bit-banging or by reconfiguring the RX0/TX0 (D0/D1) UART as SPI (but you
will loose serial I/O to the PC),

- To support 32kHz, you can try to:
- tell the compiler to optimize for speed instead for size
- try different SD cards. Some might perform better than others
- use latest sdfat library (maybe give a try to latest sdfat beta library
which might contain some performance improvements)
- fiddle around with buffer size and settings of TMRpcm and/or sdfat library
- overclock the arduino ;-)

Regards,

Marco

gamecompiler, 2017-05-23:
That "Bug" is one of the Arduino IDE. When you use PWM youll have strange output like this. The Arduino currently ("Maybe already solved in the IDE") dont makes ""real pwm"" (Theres a jump from zero to X as you can see in your graph).
Ill close this issue due this isnt a bug in this library.

closed 2017-05-23 by gamecompiler