TMRh20/TMRpcm #62: clicking sound when playing

yonoodle, 2017-03-09:
Dear contributor, 

thank you for created such good lib,  but i found some clicking sound exist during playing , 
it ticks 1 time per second , periodically (about 58 bpm )   ,the ticking sound exist only when playing , it stop as the file stop playing.   what could this issue related to?

i used arduino atmega328P 16MHz nano module , powering by computer's USB hub.

the audio file is made as the instruction indicated  ,  recording a constant pitch singing note of at least 5 second.
and i changed other file , the clicking per second

the code: 

---------------
#include <SD.h>                      
#include <TMRpcm.h>           
#include <SPI.h>

TMRpcm tmrpcm;
void setup() {
if (!SD.begin(SD_ChipSelectPin)) {return;}
tmrpcm.setVolume(5);
tmrpcm.play("me"); 
}
void loop() {
}
--------------

best regards

yonoodle

gamecompiler, 2017-03-09:
How did you interface with the DAC ? Did you try to add a low pass? (I always do, or sometimes a RC-filter, please try adding a filter and post back if you eliminate the DC)

gamecompiler, 2017-03-09:
Very noisy because of my crap soundcard (maybe also overloaded):
https://raw.githubusercontent.com/TMRh20/TMRpcm/master/audio_samples/texasradiofish_-_Slow_Cooker/recorded.ogg

Thats how the sound sounds like.

yonoodle, 2017-03-14:
what cut off frequency do you recommend?

gamecompiler, 2017-03-30:
I do not remember when id have to enter a cut off frequency Oo! Im sorry.
Did you solved your clicking noise?

Thanks in advance,
gamecompiler

closed 2017-09-27 by gamecompiler