TMRh20/TMRpcm #64: Library TMRpcm uses too much RAM

ployae, 2017-03-30:
I'm doing the project using Arduino Uno
After I uploaded #include<TMRpcm.h>, the arduino shows this result:
'Global variables use 1684 bytes (54%) of dynamic memory'
I want to have more space in arduino uno
and I use just 2 function from this library
'Audio.play("file.wav");' and 'Audio.volume(1);'
so the question is can you suggest me to delete some files in the library that doesn't need in this project?

thank you

gamecompiler, 2017-03-30:
Please wait a little bit. Im writing a lot of exams this and next month. Maybe TMRh20 will take a look. 
I just wanted to show you that id read your question.

Im sorry,
gamecompiler

gamecompiler, 2017-09-27:
Id took a look over the lib. You cant. Its really well optimized. Try to optimize your code overall if you want to have more memory.

closed 2017-09-27 by gamecompiler