TMRh20/TMRpcm #69: Recording errors

Poweringg, 2017-06-07:
Hi, i tried out sound recording on my Arduino UNO R3

i used this example code: https://github.com/TMRh20/TMRpcm/tree/master/examples/xtraFeatures/Recording

i have installed all libraries (SD.h, SPI.h, TMRpcm.h), and uncommented "#define buffSize 128 and #define ENABLE_RECORDING" in pcmConfig.h

and i still getting this errors:

```

C:\Users\Powering\Desktop\sketch_may26a\sketch_may26a.ino: In function 'void loop()':

C:\Users\Powering\Desktop\sketch_may26a\sketch_may26a.ino:50:59: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 'r': audio.startRecording("test.wav",16000,A0); break;    //Record at 16khz sample rate on pin A0

                                                           ^

C:\Users\Powering\Desktop\sketch_may26a\sketch_may26a.ino:51:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 'R': audio.startRecording("test.wav",16000,A0,1); break;  //Record, but with passthrough to speaker.

                                                             ^

C:\Users\Powering\Desktop\sketch_may26a\sketch_may26a.ino:52:61: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 't': audio.startRecording("test.wav",16000,A0,2); break;  //Do not record. Output direct to speaker

                                                             ^

C:\Users\Powering\Desktop\sketch_may26a\sketch_may26a.ino:55:49: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 's': audio.stopRecording("test.wav"); break;              //Stop recording

                                                 ^

C:\Users\Powering\Desktop\sketch_may26a\sketch_may26a.ino:56:40: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         case 'p': audio.play("test.wav"); break;                       //Play the recording 

                                        ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp: In member function 'byte TMRpcm::metaInfo(boolean, char*, char*, byte)':

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1381:18: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   char* datStr = "LIST";

                  ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1382:28: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   if(infoType == 1){datStr = "ID3 "; datStr[3] = 3;}

                            ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1410:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

  char* tagNames[] = {"INAM","IART","IPRD"};

                                          ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1410:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1410:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1430:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    tagNames[0] = "TPE1"; tagNames[1] ="TIT2"; tagNames[2] ="TALB";

                ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1430:38: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    tagNames[0] = "TPE1"; tagNames[1] ="TIT2"; tagNames[2] ="TALB";

                                      ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1430:59: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    tagNames[0] = "TPE1"; tagNames[1] ="TIT2"; tagNames[2] ="TALB";

                                                           ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp: In member function 'void TMRpcm::finalizeWavTemplate(char*)':

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1568:65: warning: narrowing conversion of '(fSize >> 16)' from 'long unsigned int' to 'byte {aka unsigned char}' inside { } [-Wnarrowing]

  seek(4); byte data[4] = {lowByte(fSize),highByte(fSize), fSize >> 16,fSize >> 24};

                                                                 ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1568:77: warning: narrowing conversion of '(fSize >> 24)' from 'long unsigned int' to 'byte {aka unsigned char}' inside { } [-Wnarrowing]

  seek(4); byte data[4] = {lowByte(fSize),highByte(fSize), fSize >> 16,fSize >> 24};

                                                                             ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp: In member function 'void TMRpcm::createWavTemplate(char*, unsigned int)':

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1608:5: error: 'SdVolume' was not declared in this scope

     SdVolume vol;

     ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1615:13: error: 'class Sd2Card' has no member named 'init'

  if (!card1.init(SPI_FULL_SPEED,CSPin)) {

             ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1619:6: error: 'vol' was not declared in this scope

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

      ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1620:21: error: 'vol' was not declared in this scope

  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }

                     ^

C:\Users\Powering\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1623:46: error: 'BLOCK_COUNT' was not declared in this scope

  if (!fil.createContiguous(&rut, fNam, 512UL*BLOCK_COUNT)) {

                                              ^

exit status 1
```

What should i do? thanks for answer.

hazee007, 2018-10-13:
in pcmConfig.h also uncomment  #define BLOCK_COUNT

amerinoo, 2019-04-02:
In order to resolve this error @Poweringg  you need to uncomment these three lines in the pcmConfig.h file located in the TMRpcm arduino library folder.
- #define buffSize 128 
- #define ENABLE_RECORDING"
- #define BLOCK_COUNT

Hope this helps!

zzmbird, 2019-12-05:
@amerinoo this cant help me.What else can i do?

amerinoo, 2019-12-05:
Why not? What error do you have? Have you uncomment these three lines? Please give more information in order to help you.

zzmbird, 2019-12-07:
yes.i do uncomment these:
#define buffSize 128
#define ENABLE_RECORDING"
#define BLOCK_COUNT
but it didnt work.

here are my code:

#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>
#define SD_ChipSelectPin 10  
TMRpcm audio;   
void setup() {
  audio.speakerPin = 4; 
  Serial.begin(115200); 
  if (!SD.begin(SD_ChipSelectPin)) {  
    Serial.println("SD Fail");
    return;
  }else{
    Serial.println("SD OK"); 
  }
  audio.CSPin = SD_ChipSelectPin;
}
void loop() {
    if(Serial.available()){                          
      char c = Serial.read();
      Serial.println(c);
      switch(c){
        case 'r': audio.startRecording("test.wav",16000,A0); break;  
        case 's': audio.stopRecording("test.wav"); break;         
      }
    }
}

i got those errors:

Arduino:1.8.10 (Mac OS X), 开发板："Arduino/Genuino Uno"

/Users/apple/Documents/Arduino/sketch_dec07b/sketch_dec07b.ino: In function 'void loop()':
sketch_dec07b:30:25: error: 'class TMRpcm' has no member named 'startRecording'
         case 'r': audio.startRecording("test.wav",16000,A0); break;   
                         ^~~~~~~~~~~~~~
sketch_dec07b:31:25: error: 'class TMRpcm' has no member named 'stopRecording'
         case 's': audio.stopRecording("test.wav"); break;                 
                         ^~~~~~~~~~~~~
"TMRpcm.h" 对应多个库
 已使用： /Users/apple/Documents/Arduino/libraries/TMRpcm
"SD.h" 对应多个库
 已使用： /Users/apple/Documents/Arduino/libraries/SD
 未使用：/Users/apple/Desktop/Arduino.app/Contents/Java/libraries/SD
"SPI.h" 对应多个库
 已使用： /Users/apple/Desktop/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI
exit status 1
'class TMRpcm' has no member named 'startRecording'

plz help me thanks.

amerinoo, 2019-12-07:
Hi @zzmbird , reading your log seems this issue is not related with your problem, please check issue #58, it talks about the error: 'class TMRpcm' has no member named 'startRecording'.

zzmbird, 2019-12-09:
@amerinoo  Hi,thank you for your help.But #58 cannot help me TAT.

closed 2020-05-11 by TMRh20