TMRh20/TMRpcm #71: Pop/clunk sound in the beginning of playback (and workaround in my case)

fusonix, 2017-08-28:
Hi,

Thank you for a great library!

In my setup I have this annoying pop/clunk sound in the beginning of the playback (it not in the end). 

I noticed that when doing a play() with an offset of 1 sec. fixed the problem ie. play('SOUND.WAV',1). The downside was that I lost 1 sec. in the beginning. 
FYI If I played the WAV file on my laptop/Windows it was just ok (no pop/clunk). 

So I opened one of my WAV files in a hexeditor and it seems like the metadata in the file was ending at around byte 80 (50h).

I then looked in library code (TMRpcm.cpp) and in the end of the method TMRpcm::wavInfo() there is a seek(44). I changed this to seek(80). This seems to work! No pop/clunk :-)

So it looks like the pop/clunk sound is playback of some of the metadata?

Is there something wrong with my WAV encoding? 

FYI 
- I'm using Arduino Nano (ATmega328)
- The WAV files are 8-bit unsigned WAV encoding, 16000 Hz (also 11025 Hz work but higher don't), using Audacity 2.1.5 with FFmpeg Export on Windows
- Using #define HANDLE_TAGS in the pcmConfig.h don't work

DrCultist, 2021-06-02:
THANK YOU! I had this problem for days now!

TMRh20, 2021-10-30:
I would assume it is some metadata or something with the data at the beginning of playback.

closed 2021-10-30 by TMRh20