TMRh20/TMRpcm #75: Sound is distorted when the delay() function is not used

Scaronthesky, 2017-11-23:
Hi folks, 

I'm facing a play back issue. When I use the delay function all works fine. When I don't use it, the sound is distorted. Could you please tell me why that might be the case? I've included an Adafruit NeoPixel stripe as you can see. 

`// Include libraries
#include <Adafruit_NeoPixel.h>
#include <SD.h>
#include <TMRpcm.h>

// Define pins
#define sdChipSelectPin 4
#define loudSpeakerPin 9
#define neoPixelPin 6
#define numberOfLeds 60

// Setup Adafruit NeoPixel stripe
Adafruit_NeoPixel strip = Adafruit_NeoPixel(numberOfLeds, neoPixelPin, NEO_GRB + NEO_KHZ800);

// Setup class for Wav file playback
TMRpcm tmrpcm;

void setup() {

  // Setup stripe
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'

  // Setup wav file play back
  tmrpcm.speakerPin = loudSpeakerPin;
  if (!SD.begin(sdChipSelectPin)) {
    return;
  }
  tmrpcm.volume(7);
   
}

void loop() {

  // doesn't work, sound is distorted - why?
  if (!tmrpcm.isPlaying()) {
   tmrpcm.play("6.wav");
  }

  // this works
  // tmrpcm.play("6.wav");
  // delay(1000);

}`

TMRh20, 2018-01-19:
I have no idea

Scaronthesky, 2018-01-22:
What caused the issue was that the LEDs of the Neopixel stripe were updated on every iteration of the loop method. Now, I'm only updating the Neopixel stripe when it is really necessary. I guess that controlling the Neopixel stripe and the SD card at the same time is just too much for the Arduino Nano.

closed 2018-01-22 by Scaronthesky