TMRh20/TMRpcm #76: What hardware ?

Mavromatika, 2018-01-19:
Hi,
I've been playing with your library for a little while and I think it's very useful. I'm writing my question as an "issue" because I don't know how to contact you otherwise. 

I'm using your library on a Uno. 
The problem I'm facing at the moment is a problem of hardware. I've built a small audio amp around an LM386, in which I plug the output of my Arduino's PWM pin directly.
The sound behaves in a strange way. If the volume pot of my amp is set to a low volume I get terrible distortion, which is less if the volume is set to the max. The distortion gets worse if I set the volume to a lower value in software.

So I'd like to know what you would suggest I put between my PWM pin and my speaker. I'm wondering in particular if I would need a DAC between my PWM pin and my amp. Thanks in advance for your help.

TMRh20, 2018-01-19:
Could you post your lm386 circuit? Don't need a DAC, the pwm is acting as a pseudo-DAC.

Here is an example using lm386:  https://circuitdigest.com/microcontroller-projects/arduino-audio-music-player

Mavromatika, 2018-01-20:
I'm using the standard circuit provided by the manufacturers, as shown on page 8 of this pdf : http://www.ti.com/lit/ds/symlink/lm386.pdf 
Thanks for the clarification about the DAC. 
I'm starting to think that maybe the pot in the circuit is there to fine tune the input gain, and should perhaps not be used as a volume pot (which would explain why it clips so terribly at certain positions). I'll try a few other things and will report here in case anybody has the same problem.

Uup115, 2018-05-17:
Hello,

I have the same problem.  Yes, I'm using the circuit, in the link you posted:
https://circuitdigest.com/microcontroller-projects/arduino-audio-music-player

The only difference, I'm working with the Arduino Pro Mini.  When playing your sample "music" files, using the basic sketch, I get garbled audio mixed with distortion.  Per the instructions, in the link above, I tried making my own music file.  Same issue.  However, if I add the following two lines in setup(), the file plays fairly well (no garble and distortion is noticed only during quiet spots).  Any values, other than 5 & 1, produce garble and distortion.

tmrpcm.setVolume(5);    //   0 to 7. Set volume level
tmrpcm.quality(1);        //  Set 1 for 2x oversampling Set 0 for normal

Help would be greatly appreciated, as I really need to use this code for a project.  Thanks.

Best Regards,
Aaron

jonathanrjpereira, 2018-08-02:
Most likely the Pot is used to tune the gain. Try changing the volume in software `tmrpcm.setVolume(3);` before changing the gain. In my case, it helped with the clipping.

closed 2020-05-11 by TMRh20