TMRh20/TMRpcm #77: TMRpcm SdFat Problem

guilhermedefalque, 2018-02-26:
Hello,

I am having a problem with the SDfat library in conjunction with the TMRPCM library.

The following error I encounter when I try to compile the code:

D: \ Users \ guilherme.defalque \ Desktop \ arduino-1.6.3-windows \ arduino-1.6.3 \ libraries \ TMRpcm-master \ TMRpcm.cpp: In member function 'void TMRpcm :: createWavTemplate (char *, unsigned int ) ':
D: \ Users \ guilherme.defalque \ Desktop \ arduino-1.6.3-windows \ arduino-1.6.3 \ libraries \ TMRpcm-master \ TMRpcm.cpp: 1609: 5: error: 'SdVolume' was not declared in this scope
     SdVolume vol;
     ^
D: \ Users \ guilherme.defalque \ Desktop \ arduino-1.6.3-windows \ arduino-1.6.3 \ libraries \ TMRpcm-master \ TMRpcm.cpp: 1609: 14: error: expected ';' before 'vol'
     SdVolume vol;
              ^
D: \ Users \ guilherme.defalque \ Desktop \ arduino-1.6.3-windows \ arduino-1.6.3 \ libraries \ TMRpcm-master \ TMRpcm.cpp: 1616: 13: error: 'class Sd2Card' has no member named ' init
  if (! card1.init (SPI_FULL_SPEED, CSPin)) {
             ^
D: \ Users \ guilherme.defalque \ Desktop \ arduino-1.6.3-windows \ arduino-1.6.3 \ libraries \ TMRpcm-master \ TMRpcm.cpp: 1620: 6: error: 'vol' was not declared in this scope
  if (! vol.init (& card1)) {} // Serial.println ("card failed"); }
      ^
D: \ Users \ guilherme.defalque \ Desktop \ arduino-1.6.3-windows \ arduino-1.6.3 \ libraries \ TMRpcm-master \ TMRpcm.cpp: 1621: 21: error: 'vol' was not declared in this scope
  if (! rut.openRoot (& vol)) {} // Serial.println ("openRoot failed"); }
                     ^
Compiling error.

Remembering that I did the requested procedure for the use of the library:

The line #define SDFAT MUST be uncommented in pcmConfig.h.

I'm using IDE 1.6.3 and the SdFat library from this location: https://github.com/greiman/SdFat. 

The code is the SdFat sample in the TMRpcm library. 

Code:

/* 
This example demonstrates how to use the SDFAT library for audio playback.
Read time and program space are reduced by using SDFAT directly

Requirements:
The SDFAT library must be installed. See http://code.google.com/p/sdfatlib/ 
The line #define SDFAT MUST be uncommented in pcmConfig.h

"error: 'File' has not been declared" means you need to read the above text
*/

#include <SdFat.h>
SdFat sd;

#define SD_ChipSelectPin 53  //use digital pin 4 on arduino Uno, nano etc, or can use other pins
#include <TMRpcm.h>          //  also need to include this library...
#include <SPI.h>

TMRpcm audio;   // create an object for use in this sketch

void setup(){

  audio.speakerPin = 11; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc
  pinMode(12,OUTPUT); //Pin pairs: 9,10 Mega: 5-2,6-7,11-12,46-45  
  
  Serial.begin(115200);
  
  if (!sd.begin(53, SPI_FULL_SPEED)) { return; 
  }else{ Serial.println("SD OK"); }
    
  audio.play("song.wav");
}

void loop(){  

  
  if(Serial.available()){   
    switch(Serial.read()){
      case 'p': audio.play("song.wav"); break;
      default: break;
    }
  }

}

Could someone please help me? Thank you.

TMRh20, 2018-02-27:
Have you tried the sdfat examples to make sure the lib is installed correctly?

guilhermedefalque, 2018-02-27:
Thanks for the answer! Yes! I was able to successfully execute all the examples from the SdFat library (2017). After much searching,  i was able to run the TMRpcm library with SdFat, but with a very old (2013) version, which no longer exists in the https://github.com/greiman/SdFat repository. Maybe, in the new versions of SdFat, some modules have changed a lot and are no longer compatible with TMRpcm. Am I right about this?

TMRh20, 2018-02-27:
I’m not sure, I deleted and then upgraded to sdfat 1.0.5 with Arduino library manager and it still works for me.

guilhermedefalque, 2018-02-27:
Hello,

Thank you for the clarification. I discovered the problem. I was adding the SdFat library manually, in the Arduino folder. Adding it through the library manager, everything worked normally. Thank you.

guilhermedefalque, 2018-02-27:
For me, it still has not worked ... The same mistake still happens. I have no idea what it is ... would you have any idea what might be happening?

thegoodhen, 2019-04-21:
I can confirm this, no luck even with old (1.0.1) version of the SDFat library. Same error as OP.

Shubhayu-Das, 2020-01-31:
Still facing this issue where the SDFat library just doesn't seem to work with the TMRpcm library. It always fails to open the SD card for writing the data from the buffers. You should probably provide a choice to the user whether or not to use the SDFat library.

P.S.: My SD card module and the SD card are in perfect working condition. It is a compilation problem that hits the program every time I try to upload to my Arduino Uno.

TMRh20, 2020-01-31:
There is a choice of sdfat or you can use the stock SD library.

Shubhayu-Das, 2020-02-01:
Ok, I found the flag in pcmConfig.h . Thank you

ShotDeath, 2020-10-19:
Hello, you have example sketch for SDfat libries?

TMRh20, 2021-10-30:
SdFat compatibility is fixed in latest release which should be available in library manager in 24hrs

closed 2021-10-30 by TMRh20