TMRh20/TMRpcm #79: Disable/Enable pseudo-DAC?

TJWi, 2018-05-25:
First of all, impressive Library.
I can see this is a great library but it might to too heavy for some projects to use it if they need to process something else, so is it possible to add a new feature that would allow user to Enable to or disable the  pseudo-DAC and possibly reduce the load it takes up by using an external DAC either R2R ladder or SPI DAC?

(I'm a beginner in coding.)
I might be completely ignorant to how much work that would be to implement it but even after reading through the TMRpcm.cpp I do not understand how the pseudo-DAC is generated and how the multitrack works.

(working on a budget project right now but right now and i might end up needed to use 2x ATmega328p, one to run the audio with this lib base on the input and another to run other program.[space constrain])

TMRh20, 2018-05-26:
Thanks, the timers etc are still required if using a dac, so I don’t know how much of a performance increase there would be. Audio is inherently cpu intensive on Arduino.
I remember somebody else emailed me saying theydid this, but don’t know if code is shared on github etc

TJWi, 2018-05-27:
Ic, thank for the reply btw, I when ahead and re-read the code and also some online explanation hopefully I can get some idea and type my own code and try. 

So I want to know if I have the correct understanding of how audio is being output by the ATmega chip. Basically a timer is setup to reset at a certain time, and an interrupt is being call when the reset happen (example: an interrupt will occur every 45us to generate a 22kHz sound) and that interrupted program will send an output signal thus creating the audio we hear. am I correct or just partially?

TMRh20, 2018-05-27:
Yes, that’s pretty much it and the pwm signal generated is the audio.

closed 2020-05-11 by TMRh20