TMRh20/TMRpcm #81: Get current time?

vin-ni, 2018-06-11:
Is it possible to get the current time value?

Thanks!

TMRh20, 2018-06-12:
> Like the audio.currentTime in javascript.
> I'm trying to switch between two audios via a button, but keep the time going:
>
> Play audio1 from 0:00 on
> Hit a Button at eg. 0:15
> Play audio2 from 0:15

Nope, that would have to be done separately.

vin-ni, 2018-06-12:
Ok,
can you give me a hint on how to achieve it? I'm really new to this.
I looked into the `pcmRF.h` but couldn't find out how to access `TIMER1`. Would that be the right way to go? Or should I start another timer that always runs when audio is playing

TMRh20, 2018-06-12:
Probably easiest to set another timer to run slowly and count the seconds/time manually

vin-ni, 2018-08-03:
How exactly would you use mili()? I ended up using an additional timer library.

closed 2020-05-11 by TMRh20