TMRh20/TMRpcm #87: Volume too low

ifeghali, 2018-08-23:
Hello 

I’m using this module as an audio amplifier 

https://goo.gl/images/ZK47qJ

But volume output from TMRpcm is too low. I can call setVolume to a maximum of 4, after that no sound can be heard. 

Exact some circuit and setup the tone() function produces audio much much louder. Any tips ?

ecoulson, 2019-05-29:
Same issue here, did you find anything?

ifeghali, 2019-05-30:
Unfortunately not.

periode, 2020-04-28:
Hey, I'm having the same problem.

@ifeghali  Did you try with another amplifier module?

ifeghali, 2020-04-28:
Yes I did, same problem. I gave up eventually, not using any circuit with
sound capabilities at the moment.

periode, 2021-06-16:
Hey @ifeghali 

I had the same issue and I changed the power supply. The Arduino Nano I'm using supplies around 4V and by using a 9V battery or a step up voltage converter to power the LM386 amp, you notice a significant difference (the LM386 can handle up to 12V).

closed 2022-12-09 by TMRh20