TMRh20/TMRpcm #90: Using functions/variables for recording audio names

FyodorKrasniy, 2018-09-23:
How does startRecording take string arguments?

Int number = 1;

 Im case of startRecording(String(number) + ".wav", 16000, A0);

Ive not been able to put in any variables into the brackets. It just keeps saying that it takes a char pointer.. 

Whats the correct way of going about this? Otherwise the ammount of files that can be generated seems limited by the naming convention

TMRh20, 2018-09-23:
It takes a character array instead of a string object.

Something like this may help:

```
  char filename[] = "music00.wav";

...

  tmrpcm.startRecording(filename,16000,A0);
  if(filename[6] == 9){
    filename[6] = 0;
    filename[5]++;
  }else{
    filename [6]++;
  }
```

luxarts, 2019-02-10:
Probably you solved the problem at this time but I'll leave the solution for those that are stuck with the same problem. As  TMRh20 said, it takes  char array (which is a string in C) so if you want to use a int as the name file you have to convert the number to string using the function itoa(int number, char *str, int base).
```C++
int number = 12345;
char intConverted[7]; //6 char (-32767 to 32767) + \0 char (end of string)
char filename[11]; //6 char of int + 4 char of extension (.wav) + \0 char (end of string)

itoa(number, intConverted, 10); //Convert the int into a string and save it in intConverted
strcpy(filename, intConverted); //Copy the intConverted string into the filename string
strcat(filename, ".wav"); //Add .wav to the end of the filename string

tmrpcm.startRecording(filename,16000,A0);
```

mariomeraz, 2020-01-17:
@luxarts Like a boos, u saved me

closed 2020-05-11 by TMRh20