TMRh20/TMRpcm #95: Questions about recording function

skylermbang, 2019-03-19:
First of all thanks for the all the works in this library. 

Hi. I am very new to Arduino and I am here to seek for help.

I am currently working on a recording machine that record when it reaches some sound level and record for 10 seconds and saves it in SD card. 

I have manage to save the first file in SD Card but after first recording, the sensor value become 0 so it can not detect the sound level anymore. 

I think I have a problem in the loop but can't figure it out.

Actually this question has been posted in 

https://forum.arduino.cc/index.php?topic=604649.0

and one of the reply said this is bug from the library but i didnt understand whats the meaning so here i am posting here again 

thank you so much to read 

`#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>
#define SD_ChipSelectPin 10
TMRpcm audio;
int audiofile = 0;
unsigned long i = 0;
bool recmode = 1;
const int sampleWindow = 50; // Sample window width in mS (50 mS = 20Hz)
unsigned int sample;
int Reset = 4;

void setup()
{
  Serial.begin(9600);
  pinMode(A0, INPUT);
  digitalWrite(Reset, HIGH);
  pinMode(Reset, OUTPUT);
  //pinMode(6, OUTPUT);
  //pinMode(2, INPUT_PULLUP);
  //attachInterrupt(0, button, LOW);
  SD.begin(SD_ChipSelectPin);
  audio.CSPin = SD_ChipSelectPin;
}

void loop()
{
  unsigned long startMillis = millis(); // Start of sample window
  unsigned int peakToPeak = 0;   // peak-to-peak level

  unsigned int signalMax = 0;
  unsigned int signalMin = 1024;

  // collect data for 50 mS
  while (millis() - startMillis < sampleWindow)
  {
    sample = analogRead(0);
    if (sample < 1024)  // toss out spurious readings
    {
      if (sample > signalMax)
      {
        signalMax = sample;  // save just the max levels
      }
      else if (sample < signalMin)
      {
        signalMin = sample;  // save just the min levels
      }
    }
  }
  peakToPeak = signalMax - signalMin;  // max - min = peak-peak amplitude
  double volts = (peakToPeak * 5.0) / 1024;  // convert to volts

  Serial.println(volts);
  //Serial.println(i);

  if (volts > 0.3) {

    audio.startRecording("1.wav", 16000, A0);
    recmode == 1;
    delay(10000);
    audio.stopRecording("1.wav");
    recmode == 0;
    Serial.print('Y') ;
    delay(10000);
  }
}`

TMRh20, 2019-03-20:
From the thread you posted, it looks like you got it working. I’ll have to look at this closer when/if I get a chance. Thanks.

andraugust, 2020-03-14:
I also had this same issue.  The solution on the arduino forum worked for me as well:
https://forum.arduino.cc/index.php?topic=604649.0

closed 2020-05-11 by TMRh20