TMRh20/TMRpcm #97: A question regarding playback w/o using serial portal

HaoJuUESTC, 2019-03-28:
Thanks a million for all your work in this project!!!

Well, I came across a strange problem... The system works just fine when I ran the code as is attached below, playing the wav file after detecting an input from the serial. But when I commented out the detecting serial inputs part (to be short, merely having the sentence

audio.play("9.wav"); 

in the loop function), the loudspeaker stops working... Does the playback function have to rely on serial input?

Here is my code:
#include <SD.h>
#define SD_ChipSelectPin 10
 /*(I'm using an arduino pro mini, and I already commented out the #define DISABLE_SPEAKER2 line in the configuration file)*/
#include <TMRpcm.h>
#include <SPI.h>
#include "pitches.h"
#include "pcmConfig.h"

TMRpcm tmrpcm;
TMRpcm audio; 

void setup(){

audio.speakerPin = 9;
Serial.begin(9600);
if (!SD.begin(SD_ChipSelectPin))
{
      tone(9,NOTE_D4);
    delay(400);
      noTone(9);
        tone(9,NOTE_D4);
    delay(400);
      noTone(9);
        tone(9,NOTE_D4);
    delay(400);
      noTone(9);
Serial.println("SD fail");
return;
}
else
{
  Serial.println("SD Ready");
    tone(9,NOTE_D4);
    delay(400);
    tone(9,NOTE_G4);
    delay(400);
    tone(9,NOTE_A4);
    delay(400);   
    tone(9,NOTE_C5);
    delay(400);    
    tone(9,NOTE_A4);
    delay(400);    
    tone(9,NOTE_G4);
    delay(400);
    noTone(9);
  }

audio.setVolume(5);
audio.CSPin = SD_ChipSelectPin;

}

void loop(){  

   if(Serial.available()){    
    if(Serial.read() == 'p')
      { //send the letter p over the serial monitor to start playback
      audio.play("9.wav");
      }
   }
}

Thanks for your kind help in advance!

TMRh20, 2019-03-28:
You need to give the audio time to play, before the loop loops and it starts at the beginning. Try using a delay or adding something like if(!audio.isPlaying){

HaoJuUESTC, 2019-03-29:
Yay, problem solved! Thanks XD

HaoJuUESTC, 2019-03-29:
I'm really sorry for bothering again...

I'm now using the library to record and then play the very same piece of recording. However the recordin, is always about 1.5 times faster than the original song I played... The pitches are right, but the durations are much shorter (So I suppose it's not the problem with the sampling process... right?). I wonder what might be the cause, and what would be the solution?

Thanks!!!

-----------------------------------------------
My codes went as follows

#include <pcmConfig.h>

#include <SD.h>
#define SD_ChipSelectPin 10
#include <TMRpcm.h>
#include <SPI.h>
#include "pitches.h"

#define Pin_Record 3

//TMRpcm tmrpcm;
TMRpcm audio; 

volatile byte FLAG_PLAY = LOW;
volatile byte FLAG_RECORD = LOW;

void Flag_PlaytoHigh( )
{
  FLAG_PLAY = HIGH;
}
void Flag_RecordtoHigh( )
{
  FLAG_RECORD = HIGH;
}

void setup(){
attachInterrupt(digitalPinToInterrupt(Pin_Record), Flag_RecordtoHigh, CHANGE);
  audio.speakerPin = 9;
Serial.begin(9600);
if (!SD.begin(SD_ChipSelectPin))
{
      tone(9,NOTE_D4);
    delay(400);
      noTone(9);
        tone(9,NOTE_D4);
    delay(400);
      noTone(9);
        tone(9,NOTE_D4);
    delay(400);
      noTone(9);
Serial.println("SD fail");
return;
}
else
{
  Serial.println("SD Ready");
    tone(9,NOTE_D4);
    delay(400);
    tone(9,NOTE_G4);
    delay(400);
    tone(9,NOTE_A4);
    delay(400);   
    tone(9,NOTE_C5);
    delay(400);    
    tone(9,NOTE_A4);
    delay(400);    
    tone(9,NOTE_G4);
    delay(400);
    noTone(9);
  }

audio.setVolume(5);
audio.CSPin = SD_ChipSelectPin;
//pinMode(7, OUTPUT);//for microphone power supply
}

void loop(){  

    if (FLAG_RECORD == HIGH)
{
  
  if (digitalRead(Pin_Record) == HIGH)
  {
    tone(9,NOTE_A4);
    delay(600); 
    noTone(9); 

    audio.startRecording("rec.wav",16000,A0);   //Record at 16khz sample rate on pin A0

  }
  else if (digitalRead(Pin_Record) == LOW)
  {
    
    audio.stopRecording("rec.wav");
    
    tone(9,NOTE_A4);
    delay(300); 
    noTone(9); 
    delay(300);
    if(!audio.isPlaying){
  audio.play("rec.wav");
    }
  }   
  FLAG_RECORD = LOW;
  }

TMRh20, 2019-03-29:
It’s probably dropping frames when recording, if the AVR +SD card can’t keep up. Recording barely works due to the overhead. Try a lower sample rate.

HaoJuUESTC, 2019-03-31:
Thanks! I lowered the sampling rate to 8 kHz, but the problem is still there...?

HaoJuUESTC, 2019-03-31:
I tried sampling at 8192 Hz, it seems better... much better than sampling at 16kHz. Sudden acceleration of the music still occurs now and then, but generally speaking, it works.

TMRh20, 2019-03-31:
Also using the sdfat library will increase performance. It just needs to be installed and enabled in pcmConfig.h per the wiki.

HaoJuUESTC, 2019-04-01:
Thanks for the helpful suggestions!

I'm not quite sure what happened:

After I included the SDFat library instead of the SD library, the TMRpcm library kept warning me like this:
-----------------------------------------------------------------------------------------------------
 C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp: In member function 'void TMRpcm::createWavTemplate(char*, unsigned int)':

C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1609:5: error: 'SdVolume' was not declared in this scope

     SdVolume vol;

     ^

C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1616:13: error: 'class Sd2Card' has no member named 'init'

  if (!card1.init(SPI_FULL_SPEED,CSPin)) {

             ^

C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1620:6: error: 'vol' was not declared in this scope

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

      ^

C:\Users\Hao Ju\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1621:21: error: 'vol' was not declared in this scope

  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }
--------------------------------------------------------------------------------
But when I used the SD library, it compiles alright. These errors only pop out when using the SDFat library.

My codes were almost the same as the one I posted out earlier, except that I had my head files included like this, and I commented out the #define SDFAT line in pcmconfig.h. I'm using the 1.1.0 version of TMRpcm, and the 1.0.5 version of SDFat.

The current codes begin like this:
---------------------------
#include "SdFat.h"
SdFat SD;
#define SD_CS_PIN SS

#include <pcmConfig.h>

//#include <SD.h>
//#define SD_ChipSelectPin 10
#include <TMRpcm.h>
#include <SPI.h>
#include "pitches.h"
-----------------------

The rest were the same as the previous version.

Is there anything I have missed? Or should I add sth in the TMRpcm.cpp file?

Thanks >_< !

TMRh20, 2019-04-01:
It seems like you have it right. Maybe try updating sdfat library to latest? Also see if it works using the sdfat example.

HaoJuUESTC, 2019-04-01:
Thanks! I updated to SDFat ver 1.1.0. The latest SDFat examples themselves work, but the errors in my codes were still the same...

TMRh20, 2019-04-01:
Ok I think something changed with the sdfat library so I’ll have to take a look at this to fix it. Looks like you are stuck with the stock SD lib for now. :(

HaoJuUESTC, 2019-04-01:
Well, at least the current library could work (though not perfectly). Thanks a lot for all your time and efforts developing this library and your kind help answering my questions!

martincornu, 2021-06-15:
Hi @HaoJuUESTC ,

Did you find a solution to your problem?
I have the same problem: sudden accelerations that rarely occur.

Regards,
marty

TMRh20, 2021-10-30:
SdFat library compatibility is fixed in latest release which should be available in Arduino library manager in 24hrs

closed 2021-10-30 by TMRh20