TMRh20/TMRpcm #99: Question about using an Array for Playing Songs

SchnoppDog, 2019-04-10:
Hi,
I have a similiar Question a closed question from User    "simon11111".
I want to use the `tmrpcm.play()` function like this (code shortened):

````    #include <TMRpcm.h>
    TMRpcm tmrpcm;
    char *sdCardSongsVIP[] = {"music1.wav", "music2.wav", "music3.wav", "music4.wav", 
    "music5.wav", "music6.wav"};
    char *sdCardSongsUser[] = {"music1.wav", "music2.wav", "music3.wav"};
    void loop{
    ...    //Some stuff happens above here 
    tmrpcm.play(sdCardSongsUser[li]);    //li is a counter pointing to the array's position
    ...    //Some stuff happens down here 
    }    
```` 

But when I want to check for errors I get this:

```` 
libraries\TMRpcm\TMRpcm.cpp.o (symbol from plugin): In function `optionByte':

(.text+0x0): multiple definition of `a'

sketch\Code_With_RFID.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/users/jm90bari/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Compile error for board Arduino / Genuino Mega or Mega 2560.
```` 

Do you know that I did wrong?
PS: Sorry that I don't use the code tags but they aren't displayed correctly in my opinion so I displayed it in another way.

TMRh20, 2019-04-10:
As it says in the error message, the variable a has been defined twice. The error has nothing really to do with the library.

SchnoppDog, 2019-04-15:
Thank you I figured it out in my code. Now it's working.

closed 2019-05-24 by TMRh20