TMRh20/TMRpcm #100: Very suspicious code in TMRpcm.cpp

thegoodhen, 2019-05-23:
if (!card1.erase(bgnBlock, endBlock)) //Serial.println("card.erase failed");

	rut.close();
        fil.close();

On lines 1631-1634 in TMRpcm.cpp

This makes the "rut" close only when card1.erase fails. Is this the intentional behavior?

TMRh20, 2019-05-24:
I think so, it works, so it must be intentional.

thegoodhen, 2019-05-28:
Let me rephrase that. You have a bug in your code. I am pretty sure it wasn't intentional.

TMRh20, 2019-05-29:
Lol I guess I’ll need to hook up some stuff and test this out. It will take some time.

thegoodhen, 2019-05-30:
Well I can confirm it works when I change

if (!card1.erase(bgnBlock, endBlock)) //Serial.println("card.erase failed");

to

if (!card1.erase(bgnBlock, endBlock)) Serial.println("card.erase failed");

(Which effectively causes the rut.close() to run no matter what).

TMRh20, 2019-05-30:
Fixed, thanks.

closed 2019-05-30 by TMRh20

TMRh20, 2019-05-31:
Awesome, I’ll trust your testing and make the changes.