TMRh20/TMRpcm #102: Pause for different tracks

lxrd22, 2019-07-17:
Hello,
Can u please update the library so it is possible to pause the track on one of the pins?
Is it possible at all? Also you could just suggest how I can do it myself.
Thank you in advance.

TMRh20, 2019-07-18:
I think if you created a variable like isPaused, you could use it to prevent the sample counter from incrementing during playback which should basically pause playback. It should be relatively easy to do.

lxrd22, 2019-07-22:
Hello,
Actually I realized I need to just mute tracks. I defined speakerpin and speakerpin2 and they play,mute and pause together just fine. However I need to mute them separately and i faced some issues. I tried using .setVolume(0,1) and .setVolume(0,0) ,however the execution seems to mute both of them. P.S. I'm playing different wav files.
Thank you very much again.

closed 2020-05-11 by TMRh20