TMRh20/TMRpcm #106: pyBug.ino:72:44: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

doganKayadelen, 2019-08-30:
My code:
`#include <pcmConfig.h>
#include <pcmRF.h>
#include <TMRpcm.h>

/*
Steps:
1. Edit pcmConfig.h
    a: On Uno or non-mega boards, #define buffSize 128. May need to increase.
    b: Uncomment #define ENABLE_RECORDING and #define BLOCK_COUNT 10000UL

2. Usage is as below. See https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features#wiki-recording-audio for
   additional informaiton.
*/

#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>
#define SD_ChipSelectPin 4
TMRpcm audio;
char audiofile = 0;
unsigned long i = 0;
bool recmode = 0;

void setup() {
  pinMode(A0, INPUT);
  pinMode(6, OUTPUT);
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(0, button, LOW);
  SD.begin(SD_ChipSelectPin);
  audio.CSPin = SD_ChipSelectPin;

}

void loop() {
}

void button() {
  while (i < 300000) {
    i++;
  }
  i = 0;
  if (recmode == 0) {
    recmode = 1;
    audiofile++;
    digitalWrite(6, HIGH);
    switch (audiofile) {
      case 1: audio.startRecording("1.wav", 16000, A0); break;
      case 2: audio.startRecording("2.wav", 16000, A0); break;
      case 3: audio.startRecording("3.wav", 16000, A0); break;
      case 4: audio.startRecording("4.wav", 16000, A0); break;
      case 5: audio.startRecording("5.wav", 16000, A0); break;
      case 6: audio.startRecording("6.wav", 16000, A0); break;
      case 7: audio.startRecording("7.wav", 16000, A0); break;
      case 8: audio.startRecording("8.wav", 16000, A0); break;
      case 9: audio.startRecording("9.wav", 16000, A0); break;
      case 10: audio.startRecording("10.wav", 16000, A0); break;
    }
  }
  else {
    recmode = 0;
    digitalWrite(6, LOW);
    switch (audiofile) {
      case 1: audio.stopRecording("1.wav"); break;
      case 2: audio.stopRecording("2.wav"); break;
      case 3: audio.stopRecording("3.wav"); break;
      case 4: audio.stopRecording("4.wav"); break;
      case 5: audio.stopRecording("5.wav"); break;
      case 6: audio.stopRecording("6.wav"); break;
      case 7: audio.stopRecording("7.wav"); break;
      case 8: audio.stopRecording("8.wav"); break;
      case 9: audio.stopRecording("9.wav"); break;
      case 10: audio.stopRecording("10.wav"); break;
    }
  }
}`

It gives warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings] for each case. It does not save to the sd card. How can i solve it?

TMRh20, 2019-08-31:
The warning shouldn’t matter. I would double check to make sure your as card is functional and has an adequate power supply to handle writing at high speed.

doganKayadelen, 2019-09-02:
> The warning shouldn’t matter. I would double check to make sure your as card is functional and has an adequate power supply to handle writing at high speed.
> […](#)

My card is functional. The power supply is provided from the usb. I tried from another computer also but it didn't work.

TMRh20, 2019-09-03:
Is the card empty? It’s possible that it cannot allocate enough contiguous space.

closed 2020-05-11 by TMRh20