TMRh20/TMRpcm #109: Not compiling for Arduino Nano Every

Nierls, 2019-09-30:
Whenever I try to compile a file for the Arduino Nano Every which includes this library it won't compile, this code:
`#include <TMRpcm.h> `
gives this output:
`/home/builder/opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -hardware /home/builder/opt/arduino-builder/hardware -hardware /home/builder/.arduino15/packages -tools /home/builder/opt/arduino-builder/tools -tools /home/builder/.arduino15/packages -built-in-libraries /home/builder/opt/libraries/latest -logger humantags -fqbn arduino:megaavr:nona4809:mode=on -build-cache /tmp -build-path /tmp/869552230/build -verbose -prefs runtime.tools.core2-32-poky-linux.path=/home/builder/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0 -prefs runtime.tools.openocd.path=/home/builder/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs runtime.tools.CMSIS-Atmel.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs runtime.tools.arduinoOTA.path=/home/builder/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs runtime.tools.dfu-util.path=/home/builder/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs runtime.tools.CMSIS.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs runtime.tools.esptool.path=/home/builder/.arduino15/packages/esp8266/tools/esptool/2.5.0-3-20ed2b9 -prefs runtime.tools.mkspiffs.path=/home/builder/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs runtime.tools.arc-elf32.path=/home/builder/.arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs runtime.tools.i586-poky-linux-uclibc.path=/home/builder/.arduino15/packages/Intel/tools/i586-poky-linux-uclibc/1.6.2+1.0 -prefs runtime.tools.sketchUploader.path=/home/builder/.arduino15/packages/Intel/tools/sketchUploader/1.6.2+1.0 -prefs runtime.tools.avr-gcc.path=/home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs runtime.tools.arm-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-linux-gcc/4.9.3 -prefs runtime.tools.linuxuploader.path=/home/builder/.arduino15/packages/arduino/tools/linuxuploader/1.5.1 -prefs runtime.tools.xtensa-lx106-elf-gcc.path=/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs runtime.tools.bossac.path=/home/builder/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs runtime.tools.arduino101load.path=/home/builder/.arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs runtime.tools.arm-none-eabi-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs runtime.tools.x86-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/x86-linux-gcc/7.2.0 -prefs runtime.tools.flashpack.path=/home/builder/.arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs runtime.tools.nrf5x-cl-tools.path=/home/builder/.arduino15/packages/arduino/tools/nrf5x-cl-tools/9.3.1 -prefs runtime.tools.avrdude.path=/home/builder/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino16 -libraries /tmp/869552230/custom -libraries /tmp/869552230/pinned /tmp/869552230/sketch_sep30a

WARNING: library tmrpcm_1_1_0 claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (megaavr) architecture(s).

Using library sd_1_2_3 at version 1.2.3 in folder: /home/builder/opt/libraries/latest/sd_1_2_3

Using library SPI at version 1.0 in folder: /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/libraries/SPI

In file included from /home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/io.h:99:0,

from /home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/pgmspace.h:90,

from /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/cores/arduino/api/String.h:30,

from /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/cores/arduino/api/Print.h:24,

from /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/cores/arduino/api/Stream.h:25,

from /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/cores/arduino/api/Client.h:22,

from /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/cores/arduino/api/ArduinoAPI.h:29,

from /home/builder/.arduino15/packages/arduino/hardware/megaavr/1.8.4/cores/arduino/Arduino.h:23,

from /home/builder/opt/libraries/latest/tmrpcm_1_1_0/pcmConfig.h:20,

from /home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:5:

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:20:27: error: 'ICIE1' was not declared in this scope

const byte togByte = _BV(ICIE1); //Get the value for toggling the buffer interrupt on/off

^

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:103:30: error: 'TIMSK1' was not declared in this scope

volatile byte *TIMSK[] = {&TIMSK1};

^~~~~~

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:103:30: note: suggested alternative: 'TIMSK'

volatile byte *TIMSK[] = {&TIMSK1};

^~~~~~

TIMSK

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:104:31: error: 'TCCR1A' was not declared in this scope

volatile byte *TCCRnA[] = {&TCCR1A};

^~~~~~

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:104:31: note: suggested alternative: 'TCCRnA'

volatile byte *TCCRnA[] = {&TCCR1A};

^~~~~~

TCCRnA

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:105:31: error: 'TCCR1B' was not declared in this scope

volatile byte *TCCRnB[] = {&TCCR1B};

^~~~~~
[... 255 lines cut ...]
/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:540:34: error: 'TOIE1' was not declared in this scope

*TIMSK[tt] = ( togByte | _BV(TOIE1) );

^

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp: In function 'void TIMER1_CAPT_vect()':

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:596:38: error: 'TOIE1' was not declared in this scope

*TIMSK[tt] &= ~( togByte | _BV(TOIE1) );

^

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp: In member function 'void TMRpcm::disable()':

/home/builder/opt/libraries/latest/tmrpcm_1_1_0/TMRpcm.cpp:710:33: error: 'TOIE1' was not declared in this scope

*TIMSK[tt] &= ~( togByte | _BV(TOIE1) );

^

exit status 1`

Is TMRpcm simply not supported for the Arduino Nano Every, or is there some steps I could take?

TMRh20, 2019-09-30:
It’s (atmega4809) just not supported.

kaelhem, 2019-12-08:
Is there any alternative ? Would TMRpcm plan to support these kind of boards one day ?

TMRh20, 2019-12-08:
No alternative that I know of. Also TMRpcm will only support AVR devices. Possibly my AAAudio library could support it but no current plans to do so.

kaelhem, 2019-12-08:
Thanks for your quick reply.
Looks like I should have done some tests before soldering my board :D
Anyway, your library rocks with compatible boards, thank you for that!

closed 2020-05-11 by TMRh20

pecancake, 2021-05-30:
I bought a Nano Every because my original Nano didn't have enough memory for my project but I didn't know about the different architecture problem so now I'm stuck without a library to play my audio with :(

@kaelhem, if you ended up finding a solution to this problem, could you share it with me? I've been researching for quite a while trying to fix this but I feel like I've hit a dead end.

@TMRh20 realistically, how hard would it be to port this to megaAVR architecture myself? Would it be a matter of changing a few lines of code or would it require a whole rewrite...

TMRh20, 2021-06-02:
Full rewrite

kaelhem, 2021-06-15:
@pecancake For my project I finally used this library with a standard Nano board and an SD card module. It was enough for my needs. I also tried a DFPlayer Mini Module... but the sound was awful (maybe the quality of my DFPlayer...).
Good luck !

RMCybernetics, 2021-09-27:
> Full rewrite
> […](#)

Is this something you have plans to implement, or would be willing to?

TMRh20, 2021-09-27:
Nope. No plans to support any other devices in TMRpcm.