TMRh20/TMRpcm #112: Save audio to SD card via A0 and send via nRF24L01 - Not possible?

alkuentrus, 2019-11-09:
Hello again, 

I managed to make work both sketches  recording.ino and musicStreamerComplex.ino separatly .

However for my project, i would like to be able to record a message and then send it via RF.  I tried several ways to do it but no success. It should be working? See the attached code please.

`#include <SD.h>                // need to include the SD library
#define SD_ChipSelectPin 4    //example uses hardware SS pin 53 on Mega2560
#include <TMRpcm.h>            //  also need to include this library...
#include <SPI.h>
#include <RF24.h>
#include "printf.h"

RF24 radio(7, 8);                  // I use 48,49 on Mega 2560, 7,8 on Uno, Nano, etc
TMRpcm audio;
pcmRF rfAudio(radio);

void setup() {

  audio.speakerPin = 9; // Use pin 9 on 328 boards (Uno, etc)
  pinMode(10, OUTPUT);   // Pin 10 on 328 boards (Uno, etc)

  Serial.begin(115200);

  if (!SD.begin(SD_ChipSelectPin)) { // see if the card is present and can be initialized:
    Serial.println("SD fail");
    return;                          // don't do anything more if not
  } else {
    Serial.println("SD ok");
  }

  printf_begin();                  // Comment this out to save spacecaaa
  rfAudio.begin();
radio.setPALevel(RF24_PA_MIN);
  radio.setDataRate(RF24_250KBPS);

  radio.printDetails();            // Comment this out to save space

  audio.CSPin = SD_ChipSelectPin;
}

void loop(){  

  if(Serial.available()){    
    switch(Serial.read()){
    case 'a': audio.startRecording("avi.wav",16000,A0,1);             break;      
    case 'b': audio.stopRecording("avi.wav");                         break;  
    case 'c': audio.stopPlayback(); rfAudio.play("avi.wav",0);        break;   // Play to radio # 1 in the radio group
    case '1': rfAudio.broadcast(1);                                   break;   // Broadcast the current audio to a different radio/group
    case '2': rfAudio.broadcast(2);                                   break;   // Broadcast the current audio to a different radio/group  
    case '3': rfAudio.broadcast(255);                                 break;   // Broadcast the current audio to a different radio/group
    case 's': rfAudio.stop();                                         break;   // Stop radio playback
    case 'S': audio.stopPlayback();                                   break;   // Stop local playback
    case '=': audio.volume(1);                                        break;   // Volume up
    case '-': audio.volume(0);                                        break;   // Volume down
    default: break;
    }
  }
}`

alkuentrus, 2019-11-09:
I have doing some more tests and, as soon as I enable recording in pcmConfig.h, it stops compiling. I guess it's not going to be easy to figure a solution... :(

alkuentrus, 2019-11-12:
Been strugling with this issue. So I got redefinition of vars and vectors timer1 related. That means that there actually shared resources (is it redefining the timer1 settings twice, maybe?)....

TMRh20, 2019-11-13:
Not sure if you can enable rf transmission at the same time as recording. That might be the problem.

alkuentrus, 2019-11-13:
:) I did believe I could, but I have been reading the code, the libs, and I think that is not possible, because there are redefinitions of the same named vectors, variables, timers, etc... Would have to rewrite the whole strucuture and I'm not that master in coding :) Thank you :)

zzmbird, 2019-12-05:
Can it be possible? It said my program had no number"startrecording"

zzmbird, 2019-12-05:
Can it be possible? It said my program had no member"startrecording"

alkuentrus, 2019-12-05:
I wasn't able to do it. There are deep changes needed in code that are beyond my programmer skills..

closed 2020-05-11 by TMRh20