TMRh20/TMRpcm #114: Continue recording if sound is present

forfrends, 2019-12-13:
Good day!
Thank you for your work! The library turned out to be very useful!
I am creating a small project - voice activated voice recorder.
I wrote a code in which the recording starts automatically if the sound level exceeds a certain value. Recording stops after a specified period of time. I try to make the recording not stop, but continue if there is sound. Here is the code:
`#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>
#include <EEPROM.h>

#define SD_ChipSelectPin 10  //

#define ledStart 2         /* PIN LED Start */
#define ledStop 3          /* PIN LED Stop */
#define MicPin A1         
//#define ReadPin A2         

char NameRecord[10];    // Имя нового - записываемого файла на SD-карту
int RecordNumber;       // Номер записи - храним в EEPROM. в диапазоне от 0 до 32767
byte Recording = 0;     // Переменная определяет идет сейчас запись или нет
int RecInterval = 5;    // Минимальная продолжительность записи в секундах
unsigned long TimeInterval = 0; // Переменная для вычисления времени
int MaxAnalogPinValue = 1020;   // Уровень сигнала на аналоговом входе при котором произойдет старт записи
int SaveADC1 = 0;
int SaveADC2 = 0;

                                //

TMRpcm audio;   // create an object for use in this sketch 

void setup() {
    Serial.begin(9600);
    
    pinMode(ledStart, OUTPUT);
    pinMode(ledStop, OUTPUT);
    //pinMode(14, OUTPUT); //A0
    //pinMode(16, OUTPUT); //A2
    //digitalWrite(14, LOW);
    //digitalWrite(16, LOW);
    audio.speakerPin = 11; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc
    pinMode(10,OUTPUT);  //Pin pairs: 9,10 Mega: 5-2,6-7,11-12,46-45
    
    if (!SD.begin(SD_ChipSelectPin)) {  
        digitalWrite(ledStop, LOW);
        digitalWrite(ledStart, LOW);
        return;
    }else{
        digitalWrite(ledStop, LOW);
        digitalWrite(ledStart, HIGH);
    }
    analogReference(EXTERNAL);
    // The audio library needs to know which CS pin to use for recording
    audio.CSPin = SD_ChipSelectPin;
    
    RecordNumber = EEPROM.read(0);
    RecInterval = RecInterval * 1000;
}

void loop() {
    if (Recording == 0){
        int AnR = analogRead(MicPin);
        Serial.println(AnR);
        if (AnR >= MaxAnalogPinValue) {

            Serial.println(1);
            StartRec();
        }
    }
    
    if (Recording == 1 && millis() - TimeInterval >= RecInterval) {
        int TempSaveA = ADCSRA;
        int TempSaveB = ADCSRB;
        ADCSRA = SaveADC1;
        ADCSRB = SaveADC2;
        int AnR = analogRead(MicPin);
        Serial.println(AnR);
        if (AnR < MaxAnalogPinValue) {
            ADCSRA = TempSaveA;
            ADCSRB = TempSaveB;
            StopRec();
            Serial.println(2);
        }else{
            TimeInterval = millis()-1;
            Serial.println(3);
        }
        ADCSRA = TempSaveA;
        ADCSRB = TempSaveB;
    }
}

void StartRec() {                   // begin recording process
    SaveADC1 = ADCSRA;
    SaveADC2 = ADCSRB;
    Recording = 1;
    digitalWrite(ledStop, HIGH);
    digitalWrite(ledStart, LOW);
    RecordNumber++;
    if (RecordNumber > 32766)RecordNumber = 0;      // небольшое огриничение в номерации файлов
    EEPROM.write(0, RecordNumber);                  // Сохранение в EEPROM номера последнего аудиофайла
    TimeInterval = millis();           // Запоминаем  millis для отсчета времени записи
    sprintf(NameRecord,"%d.wav", RecordNumber);     // создаем название файла из номера и расширения ".wav"
    audio.startRecording(NameRecord, 16000, MicPin, 0);// Старт записи
}

void StopRec() {                    // stop recording process, close file
    audio.stopRecording(NameRecord);// Стоп записи
    digitalWrite(ledStop, LOW);
    digitalWrite(ledStart, HIGH);
    Recording = 0;
    ADCSRA = SaveADC1;
    ADCSRB = SaveADC2;
}`
But the recording still stops, and starts again, instead of continuing without stopping if there is sound.
Can you tell me how to fix this?
Sorry for my english.

TMRh20, 2019-12-13:
You can’t really call analogRead while recording is happening, but you might be able to get the value of the ADCH register (8-bit) and maybe average that over time to get the sound level.

Ie:
uint8_t volume = ADCH;

forfrends, 2019-12-15:
Thanks for the help!
I bring a complete sketch. Recording is activated by voice. If there is sound, then the recording is not turned off. Each new recording is saved with a new file name.
The circuit diagram of the device was created in the Freezing program: https://drive.google.com/open?id=18qedImXvcyalzkrTOmg53ig3n2Y3WLlL
`/*
This sketch demonstrates recording of standard WAV files that can be played on any device that supports WAVs. The recording
uses a single ended input from any of the analog input pins. Uses AVCC (5V) reference currently.

Requirements:
Class 4 or 6 SD Card
Audio Input Device (Microphone, etc)
Arduino Uno,Nano, Mega, etc.

Steps:
1. Edit pcmConfig.h
    a: On Uno or non-mega boards, #define buffSize 128. May need to increase.
    b: Uncomment #define ENABLE_RECORDING and #define BLOCK_COUNT 10000UL

2. Usage is as below. See <a href="https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features#wiki-recording-audio" title="https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features#wiki-recording-audio" rel="nofollow">https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features#wiki-recording-a...</a> for
   additional informaiton.

Notes: Recording will not work in Multi Mode.
Performance is very dependant on SD write speed, and memory used.
Better performance may be seen using the SdFat library. See included example for usage.
Running the Arduino from a battery or filtered power supply will reduce noise.
*/

#include <SD.h>
#include <SPI.h>
#include <TMRpcm.h>
#include <EEPROM.h>

#define ENABLE_DEBUG        //Включение вывода отладочной информации через Serial. Только для отладки! Закоментировать эту строку перед заливкой в готовое устройство!

#define SD_ChipSelectPin 10  //

#define ledStart 2         /* PIN LED Start */
#define ledStop 3          /* PIN LED Stop */
#define MicPin A1          // Аналоговый пин, к которму подключен микрофон

char NameRecord[10];        // Имя нового - записываемого файла на SD-карту
int RecordNumber;           // Номер записи - храним в EEPROM. в диапазоне от 0 до 32767
byte Recording = 0;         // Переменная определяет идет сейчас запись или нет
                            //-----------------------------------------------
int RecInterval = 5;        //! Минимальная продолжительность записи в секундах !
                            //-----------------------------------------------
unsigned long TimeInterval = 0; // Переменная для вычисления времени
int MaxAnalogPinValue = 1000;   // Уровень сигнала на аналоговом входе при котором произойдет старт записи
                                // Значение подбирается индивидуально! (100 - 1024)
int SaveADC1 = 0;
int SaveADC2 = 0;

unsigned int sample;
unsigned int signalMax = 0;
unsigned int signalMin = 256;
unsigned int peakToPeak = 0;
#define peakToPeakMinLevel 100   // Уровень срабатывания на звук. Значение подбирается индивидуально!
                                // Максимальное значение - 255

TMRpcm audio;   // create an object for use in this sketch 

void setup() {
    #if defined (ENABLE_DEBUG)
        Serial.begin(9600);
    #endif
    
    pinMode(ledStart, OUTPUT);
    pinMode(ledStop, OUTPUT);
    audio.speakerPin = 11; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc
    pinMode(10,OUTPUT);  //Pin pairs: 9,10 Mega: 5-2,6-7,11-12,46-45
    
[... 71 lines cut ...]
    }
}

void StartRec() {                   // begin recording process
    SaveADC1 = ADCSRA;
    SaveADC2 = ADCSRB;
    Recording = 1;
    digitalWrite(ledStop, HIGH);
    digitalWrite(ledStart, LOW);
    RecordNumber++;
    if (RecordNumber > 32766)RecordNumber = 0;      // небольшое огриничение в номерации файлов
    EEPROM.write(0, RecordNumber);                  // Сохранение в EEPROM номера последнего аудиофайла
    TimeInterval = millis();                        // Запоминаем  millis для отсчета времени записи
    sprintf(NameRecord,"%d.wav", RecordNumber);     // создаем название файла из номера и расширения ".wav"
    audio.startRecording(NameRecord, 16000, MicPin, 0);// Старт записи
}

void StopRec() {                    // stop recording process, close file
    audio.stopRecording(NameRecord);// Стоп записи
    digitalWrite(ledStop, LOW);
    digitalWrite(ledStart, HIGH);
    Recording = 0;
    ADCSRA = SaveADC1;
    ADCSRB = SaveADC2;
}`

closed 2019-12-15 by forfrends