TMRh20/TMRpcm #116: Recorded WAV File is Empty

ronaktal, 2019-12-18:
Hi , thank you for the wonderful library !

When attempting to record , on the Mega , we are getting an appropriate length WAV file that is empty , and is silent without any data. 

This is with the following changes in pcmConfig.h:

Uncommented:

#define ENABLE_RECORDING
#define BLOCK_COUNT 10000UL 

Default pcmConfig.h has SPI_FULL_SPEED uncommented.

Here is the relevant code we are using:

`audio.startRecording("1.wav",16000,A0);` 

When we test analogRead(A0) we do get hardware input from out MIC (which is the MAX9814 adafruit board) .

We do get a file 1.wav generated , but when inspected in Audacity each sample has the sample null value saved , as if we had the wrong analog pin. Could only certain Analog pins work for record feature ?

Thank you for your help !

Ronak

TMRh20, 2019-12-19:
Hmm, not sure what to suggest. If you are getting good readings using analogRead() then it should be getting a good input. Maybe the issue is with the SD writing? Have you tested read/write functionality? Note: Recording at 16 kHz results in a write 16,000 times/second so the SD card will pull quite a bit more power when recording vs doing some simple reads/writes to the card. Make sure you have adequate power.

ronaktal, 2019-12-19:
Thank you ! SD Write was tested to work , however it may be the write speed that is the issue , I am not sure. 

I just wanted to try to Serial output the actual pin value it is writing to the buffer, however I am not sure where/how to put a Serial.println to test this in the startRecording function ? 

I also cannot seem to find where it writes the buffer to the actual SD card , I just wanted to debug that part too to see if the card is responding properly.

Thank you for your help . I actually did get this to work when I first loaded the sketch a while ago , hasn't worked since , I suspect it's some SD / Hardware issue. 

Best,

Ronak.

TMRh20, 2019-12-19:
You can’t really Serial.print the values, it happens too fast. You can access the ADCH register while recording is happening, and print that out. Ie: uint8_t volume = ADCH;

Kind of the same with with the sd writing, it happens so fast and within an interrupt, so very difficult to debug. See https://github.com/TMRh20/TMRpcm/blob/master/TMRpcm.cpp#L566

ronaktal, 2019-12-21:
Many thanks for your help so far ! So it looks like , as you suggested , this is an SD card issue. 

When we check  sFile.available() prior to the write , it fails. However , using the ReadWrite example on the SD library , we are successfully able to write the characters "testing 1,2,3" . It seems when we try to write the ADC data from the buffer it is failing.

Any suggestions or tips appreciated , thank you again :)

ronaktal, 2019-12-21:
Hi , working now, thank you !

Turns out the following code in the Setup was causing an issue with recording to the SD card:

`audio.speakerPin = 5;`

Not sure why this is , once this is moved into the loop section , it works normally. 

Thank you , this may help someone out there so posting it as what works for me. However, if you have any suggestions as to why that would be helpful since keeping this in Setup is preferred.

mdshabaz002, 2020-01-09:
I'm trying to record a voice message in WAV format using ISD1820 (recording module) and save it to a  SD card. The issue is the file created only has noise in it. I've tried a lot which includes commenting audio.speaker pin but the result is still the same. I know the SD module connections but i guess the recording module is the culprit here.
Can you please guide me to the pin connection of your recorder module to Arduino.

ronaktal, 2020-01-23:
Hi sorry for the late reply. I has just used A15 pin , but am using the
max9814 chip. Not sure if younhave solved the problem , but I would suggest
looking at the recording module first.

Best wishes !

Ronak

closed 2020-05-11 by TMRh20

ronaktal, 2020-05-11:
Thank you for the feedback and response. We were able to get this working
and I think it had to do with a RAMP setting to prevent clicks / popping
noises . Once that was uncommented it seemed to work ok.

Any issues I will reach out but thank you again !

Ronak.