TMRh20/TMRpcm #117: WAV file not playing

MatthewAlviz, 2020-01-05:
Hello, I hope you can help me with this one. I have some audio files converted to 8-bit,16000Hz Sample rate, mono that doesn't play but there are some that works.

TMRh20, 2020-01-05:
Have you tried playing the files on PC? What are you using to convert them? Are you getting any kind of sound or just nothing at all?

mdshabaz002, 2020-01-09:
The file I'm getting contains a slow buzz noise!

derekxkwan, 2020-04-12:
i'm also experiencing problems. i've tried expoorting wav files from audacity, ffmpeg, and sox in unsigned 8 bit mono at various sample rates 16k-32k and i'm just getting quiet warbly noise. the included sample wav "music" works fine but my own exported ones don't work and struggling to figure out why, although i'm not discounting it's me doing something wrong on my end.

TMRh20, 2020-04-12:
Can you post a short clip of a problem wav file here?

derekxkwan, 2020-04-14:
thanks for looking to this, uploaded a short clip windchimes sound from freesound zipped up. exported from Audacity on Ubuntu 18.04.4 LTS.

output from ffprobe:
`Input #0, wav, from '/home/dxk/arduinosounds/windchimes.wav':
  Duration: 00:00:30.10, bitrate: 256 kb/s
    Stream #0:0: Audio: pcm_u8 ([1][0][0][0] / 0x0001), 32000 Hz, 1 channels, u8, 256 kb/s
`
[windchimes.zip](https://github.com/TMRh20/TMRpcm/files/4473327/windchimes.zip)

TMRh20, 2020-06-06:
Hmm, I replied to this comment earlier, it shows in my email but not on github? Weird... anyway:

Well the file posted appears to be fine, actually is the exact same format as the example file, so I don’t really have an idea of what the problem would be.

One possibility is that the data on Sd card is getting corrupted or read incorrectly. Beyond that I can’t see why the example would work but not your own files. Of course you could test this by printing out the first 60 or so bytes of the file from SD, and compare to the wav file viewed in a hex editor on pc.

I can’t come up with any other reasoning for this behaviour at the moment...

diego-ramirez-dbr, 2020-06-20:
Also facing the same problem, sample files on tutorials I have found work, but the samples I convert myself don't. Have used Audacity and an online converter. 

One strange thing is that one of the files does work, but when I do the same process for other clips (Audacity, mono, 16kHz, unsigned 8-bit) it doesn't play.

TMRh20, 2020-06-20:
Not sure if it will help, but @diego-ramirez-dbr could you possibly post a copy of the converted file that does work along with one that doesn't? I'd like to test again to see if I can find any differences or recreate the issue.

diego-ramirez-dbr, 2020-06-21:
Sure, santa3.wav is the one that works and so_stupid.wav is silent. Excuse the language, I'm making an Always Sunny quote generator.
[proj_audio_files.zip](https://github.com/TMRh20/TMRpcm/files/4808753/proj_audio_files.zip)

Those were both recorded/exported in Audacity 18.04.4, moved to a Windows 10 machine by flash drive, and put on an sd card from there. I've found better luck doing recording/exporting from Windows machine directly, will update with any other findings.

TMRh20, 2020-06-21:
Ok, I should have noticed this earlier, but it seems that all the problem files in question so far break the 8-character limit for file names per https://www.arduino.cc/en/Reference/SDCardNotes :

>File Naming
FAT file systems have a limitation when it comes to naming conventions. You must use the 8.3 format, so that file names look like “NAME001.EXT”, where “NAME001” is an 8 character or fewer string, and “EXT” is a 3 character extension. People commonly use the extensions .TXT and .LOG. It is possible to have a shorter file name (for example, mydata.txt, or time.log), but you cannot use longer file names. Read more on the 8.3 convention.

derekxkwan, 2020-06-21:
hey, that worked! thx for your help and also pointing out arduino things I probably should've done more research on... 👍

closed 2021-10-30 by TMRh20

DooblyNoobly, 2022-02-24:
> Ok, I should have noticed this earlier, but it seems that all the problem files in question so far break the 8-character limit for file names per https://www.arduino.cc/en/Reference/SDCardNotes :

Thank you! I spent like 2 days trying to work this problem out