TMRh20/TMRpcm #121: Best board to achieve 2x microphone at 44.1k sampling rate 16 bit-depth with TMRpcm library

Akirakato1, 2020-03-06:
Hi,
I'd like to know what the most powerful board that this library supports since I saw in wiki a warning about processor intensive and can reduce recording quality. I need two mics and wondering if your library handles simultaneous recording of two mics, i'm thinking record for each mic and merge the data to then write into the SD card. Is there way I can do that? the saved file don't have to be directly wav format, just need to be decoded later with script on computer.
Thanks

Akirakato1, 2020-03-06:
or can I just ```audio.startRecording(pin1); audio.startRecording(pin2) audio.stopRecording(pin1); audio.stopRecording(pin2);``` and have it be recording simultaneously. I'd like continuous recording over 24 hours with 32gb sd card, should be plenty, then what interval should I call "stopRecording" to achieve continuous audio data?

TMRh20, 2020-03-06:
Nope.

The most powerful board TMRpcm supports is Arduino Mega2560.

This will handle 8-bit, mono, 16-24khz recording at best.

I’ve created another library for Arduino Due which supports sampling on multiple pins, and can handle an overall datarate of 200kB/s+ depending on SD write speed. See https://github.com/TMRh20/AutoAnalogAudio
and
http://tmrh20.blogspot.com/2016/11/new-automatic-analog-audio-library-for.html?m=1

To calculate your requirements Datarate = SampleRate * Channels * BitRate

Peace,
TMRh20

Akirakato1, 2020-03-09:
Thanks for the response,
Is it also possible to choose non-byte aligned bit-depth like 12?

TMRh20, 2020-03-09:
Yeah, Due ADC is 10 or 12 bit, DAC is 12 bit.

Akirakato1, 2020-03-10:
will Arduino Micro work with this library?

TMRh20, 2020-03-10:
It should.

Akirakato1, 2020-03-11:
I uncommented #define ENABLE_RECORDING and #define BLOCK_COUNT 10000UL  // 10000 = 500MB   2000 = 100MB, and use 32GB class 4 sandisk micro sd connected using XCSOURCE 1A 5V Micro USB TP4056. I looked at the mapping here https://store.arduino.cc/usa/arduino-micro and connected gnd->gnd, vcc->3.3V, miso->miso, mosi->mosi, sck->sck, and cs->4 (digital pin). When I run your exampls I get "SD Fail" serial output. Do you have an idea what's going on?

TMRh20, 2020-03-13:
FYI this space is really for reporting technical issues or potential issues with the library and related issues, it is not for technical support.

However, I would suggest standard troubleshooting of your SD card, wiring etc.

closed 2020-05-11 by TMRh20